Ultimaker/Cura #2350: "Save to" menu selection should persist across launches

ShakeThat, 2017-08-31:
Usually I want to **Save to File**, but that is never the initial choice presented to me when I launch Cura. Instead, the initial choice is always **Save to Removable Drive** (actually, now that I've got a connected printer the initial choice is always **Print with OctoPrint**). Once selected, the choice persists until I quit the application.

I propose that the choice should be saved, and then restored when the menu is populated on launch, provided that it is still among the options available at that time.

Ghostkeeper, 2017-09-10:
Yeah, good point.

Ghostkeeper, 2017-12-04:
This was removed from our planning by our project manager because it's older than 12 weeks.