Ultimaker/Cura #2351: Can't slice correctly in Moai Printer Profile

ComicBit, 2017-08-31:
If i try to slice at 0.025 um on this profile on cura 2.6 i obtain this:
<blockquote class="imgur-embed-pub" lang="en" data-id="RVao8ki"><a href="//imgur.com/RVao8ki">View post on imgur.com</a></blockquote>

Instead in Cura 2.7 with the identical same settings i get this:
<blockquote class="imgur-embed-pub" lang="en" data-id="nljbplU"><a href="//imgur.com/nljbplU">View post on imgur.com</a></blockquote>

LipuFei, 2017-09-04:
Hi @ComicBit, could you create a project file out of your print and share with us if you don't mind? It may have something to do with the settings.

Ghostkeeper, 2017-09-10:
The only meaningful change made to the Moai in between the two versions was this one: https://github.com/Ultimaker/Cura/commit/c3dcee7f1499f751125796a50641b3cf120a4a7a#diff-ac4f35f8ba7567a9d992cd26210bfeb6

Perhaps `infill_overlap` has something to do with it?

Ghostkeeper, 2017-09-10:
> slice at 0.025 um

:open_mouth: 25 nanometres!

ComicBit, 2017-09-10:
@LipuFei there you go [https://www.dropbox.com/s/86gozjma90zg5fl/PM_2%20%28repaired%29_Hollowed.curaproject.3mf?dl=0](url)

@Ghostkeeper about 0.025 um..... LOL i'd made a mistake :D

ComicBit, 2017-09-23:
Any guess on this?

Ghostkeeper, 2017-09-28:
I tried and could reproduce the bug there too.

I don't know what's causing this and I can't think of anything done between 2.6 and 2.7 that would change it. I could perform a bisect on CuraEngine to see, but with a heavy model such as this and my computer not having enough memory to show this in Layer View that would take quite a lot of time.

ComicBit, 2017-09-28:
Let me know if i can help you in any way. My computer is pretty powerful

ianpaschal, 2018-03-02:
@ComicBit Sorry for the delay, are you still experiencing this issue on later versions of Cura?

Ghostkeeper, 2018-03-26:
Yeah, the bug still exists.

[image]

It's worth trying some of the meshfixing settings maybe. But I still don't get why this broke in version 3.0.

diegopradogesto, 2019-09-10:
This issue is somehow fixed or not happening anymore for this model:

[image]

closed 2019-09-10 by diegopradogesto