Ultimaker/Cura #2358: Ironing (Neosanding) issue

konvoj, 2017-09-01:
When neosanding is on, blobs of material on short strokes are extruded, see image:
[image]

This result was with default Cura profile, "enable ironing" on, profile "zigzag".

It seems that neosanding extrusion of 10% is counting with the standard print speed of the nozzle, but at short strokes thanks to the acceleration settings, this speed is never achieved, thus overextrusion appear.

After end of neosanding procedure the nozzle traveled to different locations of the print making several indentations in the top layer, which can be also visible in the picture above.

Source stl attached:
[Rudder.zip](https://github.com/Ultimaker/Cura/files/1270170/Rudder.zip)

konvoj, 2017-09-04:
Source of the blobs is now clear to me. Between top layer and the ironing layer there is a filament retraction/insertion step, that ends up in extruding more plastic than needed. 
Since I'm using 0% extrusion during for the ironing layer, the easy way out was to manually edit G-code, effectively deleting all filament inserts/retracts after the top layer final retraction has been done.
This is not an elegant solution and a parameter like "Retraction extra prime amount" having negative value only valid for the ironing level could solve this issue.
[Manual_ironing.zip](https://github.com/Ultimaker/Cura/files/1274888/Manual_ironing.zip)

Ghostkeeper, 2017-12-04:
Our project manager just removed this issue from our project planning because it's older than 12 weeks.

closed 2018-11-26 by Appesteijn

davispw, 2018-12-08:
Still a bug, just experienced this today.  Reopen?  Need me to provide a sample?