Ultimaker/Cura #2359: Install/Remove older versions of Cura

konvoj, 2017-09-01:
Lately i was upgrading from Cura 2.6.2 to 2.7 and it was quite confusing that now both versions are installed.

As part of the installation a dialog or a tick box shall be offered to the user saying something like "Do you want to remove older Cura versions". 
This would allow user to have only the most recent version if needed, without manual action of removing the older Cura version by hand.

Ghostkeeper, 2017-09-10:
It does. When you install Cura it should find that older versions are installed and should ask you whether you want to uninstall them.

Ghostkeeper, 2017-09-10:
[image]

modem7, 2017-09-11:
The folders for those versions are still on the system.
[image]

fieldOfView, 2017-09-11:
See https://github.com/Ultimaker/Cura/pull/1151

Ghostkeeper, 2018-03-26:
This is still a problem. The Windows uninstaller doesn't remove the configuration files.

closed 2023-06-24 by github-actions[bot]