Ultimaker/Cura #2360: Online printer safety

konvoj, 2017-09-01:
Having UM3 in the work, anyone who has Cura installed and physical access to the printer can connect to the printer without further restrictions as e.g. Admin password. 
This is a huge safety risk since nobody has overview of people with access and the number of people can be growing anytime someone easily connects to the printer, this can lead to confusion and misunderstandings and (un)intentional aborts of prints . 

If the user asking for permission to the printer is denied Cura live print window blocks the printer controls, but still show online printer camera image. This is logically another safety issue, since anyone, even denied users can see what is being printed. Denied users shall have complete blockage from using or viewing the printer.

nallath, 2017-09-01:
If someone has physical access to a printer, there is no way to prevent them from getting any form of access (eg; Just putting a USBDrive in there also allows the user to start prints).

konvoj, 2017-09-01:
This can happen even if someone had access to the printer once (with a bit of luck). Then he gains remote access and ability to control the printer.
The printer is placed in the room with limited access, and I don't want to be paranoic, but now almost anyone who is let in for whatever reason can have full control of the printer.

Second part of the issue is also concerning, since anyone in the same network may see what the printer is printing. In the situation the printer in limited access area is printing confidential stuff and anyone can see it is for me worrying.

fieldOfView, 2017-09-01:
You could put your UM3 on a separate network or even on a VPN

Ghostkeeper, 2017-09-10:
There's no guarding against physical access, of course. One security issue that I know of is that we never deprecate keys, so accessing someone's computer that has an authenticated key to some printer will give the attacker access to that printer indefinitely. It would be better to give the keys a lifetime of, say, 3 months, or at least to provide a way to invalidate keys that are known to be stolen.

We've considered whether to protect the camera image. In the end, the usefulness of being able to recognise a printer through its camera image won over the security issue of finding personal information in the camera image. The camera is only looking at the inside of the printer, so the user's room is not visible, was the argument. I don't think it was considered though that a user might be printing a giant spiked dildo and so the actual insides might be private as well...

fieldOfView, 2017-09-10:
I just printing... for a friend

ChrisTerBeke, 2017-09-27:
Do we still think this is an issue?

konvoj, 2017-09-28:
As industry standard, some kind of protection is a must (like all company PCs needs to use passwords to login). The level of protection vs. physical access to the printer is however debatable. You can close this issue for now, once i come with a better safety strategy, i will open another issue.

closed 2017-09-28 by ChrisTerBeke