Ultimaker/Cura #2361: Cura 2.7 Unable to zoom out

weithiam, 2017-09-01:
After zooming in to check layer by layer I am unable to zoom out anymore.  Other view changes still work.

Neo

Jhinta, 2017-09-01:
+1 this random happens

nickthetait, 2017-09-01:
Bizarre! After restarting the application everything returns to normal? How often would you describe this occurring?

Drayson80, 2017-09-06:
+1 
And additionally sometimes when manipulating the model, I get a very strange behavior of pan/zoom/turn so that at the end, the whole print platform is out of sight... any idea how to center the print platform again?

nallath, 2017-09-06:
We had the feature request for a while to reset the camera view to the home position. Haven't found the time to build it yet.

Ghostkeeper, 2017-09-13:
I've seen this sometimes when using the new "Zoom to mouse position" feature. But it's hard to reproduce.

Were you using the Zoom to mouse position thing?

Drayson80, 2017-09-13:
Acc. to my experience, it has the same behavior independend of with or w/o the Zoom to mouse position thing...

ianpaschal, 2018-02-27:
This is fixed in the most recent version of Cura (3.2.1).

closed 2018-02-27 by ianpaschal