Ultimaker/Cura #2362: Unable to open projects in 2.7 from 2.6

nickfrats, 2017-09-01:
Currently I'm trying to open up some of my projects I saved in version 2.6. Each time I try to open one, a failed error message comes up. Are the old project files not compatible with the new version? I find it easiest to save my settings in projects rather then have to import and export profiles, mainly because it saves my material and the settings all in one.

nallath, 2017-09-04:
Could you share your logs? Without it we can't see what's going wrong. I've just tested it on my machine and it seems to work just fine.

nickfrats, 2017-09-05:
In this portion of the log I opened two files successfully and two unsuccessfully. One successful file was from 2.7 and one was actually from 2.6 (this is the first time I've gotten any file from 2.6 to open). The two other files that failed to open were also from version 2.6. Afterwards I just closed out of Cura. Let me know if there is anything else I can provide to get this sorted out.

2017-09-05 10:38:12,576 - ERROR - UM.Logger.logException [76]: Exception: Could not deserialize container Ultimaker 3 #11
2017-09-05 10:38:12,579 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-09-05 10:38:12,580 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 204, in load
2017-09-05 10:38:12,582 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerStack.py", line 352, in deserialize
2017-09-05 10:38:12,583 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerStack.py", line 357, in deserialize
2017-09-05 10:38:12,584 - ERROR - UM.Logger.logException [80]: Exception: When trying to deserialize Ultimaker 3 #11, we received an unknown ID (Ultimaker 3_settings #24) for container
2017-09-05 10:38:12,589 - ERROR - UM.Logger.logException [76]: Exception: Could not deserialize container ultimaker3_extruder_left #11
2017-09-05 10:38:12,592 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-09-05 10:38:12,593 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 204, in load
2017-09-05 10:38:12,594 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\Settings\ExtruderStack.py", line 96, in deserialize
2017-09-05 10:38:12,596 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerStack.py", line 352, in deserialize
2017-09-05 10:38:12,597 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerStack.py", line 357, in deserialize
2017-09-05 10:38:12,598 - ERROR - UM.Logger.logException [80]: Exception: When trying to deserialize ultimaker3_extruder_left #11, we received an unknown ID (ultimaker3_extruder_left #2_settings #25) for container
2017-09-05 10:38:12,687 - INFO - cura.Settings.GlobalStack.addExtruder [81]: Extruder[ultimaker3_extruder_right #8] added to [Ultimaker 3 #8] at position [1]
2017-09-05 10:38:12,689 - WARNING - UM.Settings.ContainerStack.addMetaDataEntry [162]: Meta data with key machine was already added.
2017-09-05 10:38:12,694 - ERROR - UM.Logger.logException [76]: Exception: Could not deserialize container ultimaker3_extruder_right #11
2017-09-05 10:38:12,696 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-09-05 10:38:12,698 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 204, in load
2017-09-05 10:38:12,699 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\Settings\ExtruderStack.py", line 96, in deserialize
2017-09-05 10:38:12,701 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerStack.py", line 369, in deserialize
2017-09-05 10:38:12,702 - ERROR - UM.Logger.logException [80]: UM.Settings.ContainerStack.InvalidContainerStackError: Stack ultimaker3_extruder_right #11 does not have a definition!
2017-09-05 10:38:12,716 - INFO - cura.Settings.GlobalStack.addExtruder [81]: Extruder[ultimaker3_extruder_left #2] added to [Ultimaker 3] at position [0]
2017-09-05 10:38:12,718 - WARNING - UM.Settings.ContainerStack.addMetaDataEntry [162]: Meta data with key machine was already added.
2017-09-05 10:38:12,724 - INFO - cura.Settings.GlobalStack.addExtruder [81]: Extruder[ultimaker3_extruder_left #8] added to [Ultimaker 3 #8] at position [0]
2017-09-05 10:38:12,726 - WARNING - UM.Settings.ContainerStack.addMetaDataEntry [162]: Meta data with key machine was already added.
2017-09-05 10:38:12,733 - INFO - cura.Settings.GlobalStack.addExtruder [81]: Extruder[ultimaker3_extruder_right #2] added to [Ultimaker 3] at position [1]
2017-09-05 10:38:12,735 - WARNING - UM.Settings.ContainerStack.addMetaDataEntry [162]: Meta data with key machine was already added.
2017-09-05 10:38:12,738 - DEBUG - UM.Settings.ContainerRegistry.load [214]: Loading data into container registry took 11.914893388748169 seconds
2017-09-05 10:38:12,752 - DEBUG - UM.Controller.setActiveView [84]: Setting active view to SolidView
2017-09-05 10:38:15,338 - DEBUG - UM.Qt.Bindings.Theme.load [165]: Loading theme file: C:\Program Files\Cura 2.7\resources\themes\cura-dark\theme.json
2017-09-05 10:38:15,340 - DEBUG - UM.Qt.Bindings.Theme.load [165]: Loading theme file: C:\Program Files\Cura 2.7\resources\themes\cura\theme.json
2017-09-05 10:38:15,406 - DEBUG - UM.Qt.Bindings.Theme.load [232]: Loaded theme C:\Program Files\Cura 2.7\resources\themes\cura
2017-09-05 10:38:15,414 - DEBUG - UM.Qt.Bindings.Theme.load [232]: Loaded theme C:\Program Files\Cura 2.7\resources\themes\cura
[... cut ...]
2017-09-05 10:40:07,753 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-09-05 10:40:19,755 - DEBUG - UM.Qt.QtApplication.windowClosed [267]: Shutting down cura
2017-09-05 10:40:19,802 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [244]: Attempting to kill the engine process
2017-09-05 10:40:19,802 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [250]: Killing engine process
2017-09-05 10:40:19,802 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [253]: Engine process is killed. Received return code 1

LipuFei, 2017-09-05:
@nickfrats If you don't mind, could you share that failed project file so we can investigate? Thank you.

nickfrats, 2017-09-05:
Unfortunately I'm not able to share the files as they contain client sensitive information. I've gone back and tried opening all my old files from 2.6 and it seems some of them do open while others won't. There aren't any that don't open that I can share with you. My guess is that some of the project files are corrupted on my end. Thanks everyone for your help!

closed 2017-09-05 by nickfrats

Ghostkeeper, 2017-09-11:
> Exception: When trying to deserialize ultimaker3_extruder_right #11, we received an unknown ID (um3_bb0.4_ABS_Fast_print) for container

This is a weird error since the profile `um3_bb0.4_ABS_Fast_print` is built-in.