Ultimaker/Cura #2363: Cura 2.7 on Mac doesn't import material & Printer profiles from 2.6.x

hillct, 2017-09-01:
When I installed 2.7 for Mac (OS 10.12.6) the printer and material profiles entered in Cura 2.4 imported as expected, but none of the materials or printer profiles entered since then (2.5.x or 2.6.x) were available. Reverting back to 2.6.2 to see if those profiles can be recovered, exported in some way and manually imported.

nallath, 2017-09-04:
Could you share logging?

torque, 2017-09-04:
I'm also having trouble getting Cura 2.7 to import my custom materials from 2.6, but my printers appear to import mostly fine (though my printers are all built-ins, so that may be part of it). At some level, my troubles appear to be related to #2270 (which was ostensibly fixed in 2.7).

For context, I have a Prusa i3 MK2, which uses 1.75mm diameter filament. In 2.6.2, this printer appears to have the correct settings in the "machine settings" dialog: the material diameter shows 1.75mm. When launching Cura 2.7 for the first time, this printer is imported with 2.85mm material diameter, which causes the materials display to not show any 1.75mm diameter filaments until it has manually been fixed. I've checked, and the built-in `Cura2.7.app/Contents/Resources/resources/definitions/prusa_i3_mk2.def.json` definitely has `"material_diameter": { "default_value": 1.75 }` in its overrides.

This aside, none of my custom materials are imported automatically either. After setting the filament diameter so that the material manager will show the correct material sizes, I can import them manually, although they don't import quite correctly (for some reason the display name gets clobbered). [Here's a gist of a material file before and after import](https://gist.github.com/torque/85a570f9d8ecdaf17284567f52b4b5f2).

The log file appears to have no logging related to this kerfuffle, but I've included the full `cura.log` covering initial import through fixing the printer setting and manually importing my old materials [in this gist](https://gist.github.com/torque/91948c7f2786c728b4b414a33d23a0cb). Is there any way to get more verbose logging?

hillct, 2017-09-04:
Please find my Cura.log file fragment for review, here: https://gist.github.com/hillct/bffbe000d85f514eb9d368418f3d05f0

Another interesting item of note - it appears that the content (or a subset thereof) of the ~/Library/Cura/2.6 folder was copied into the ~/library/Cura/2.7/2.6 folder, including the original content of ~/Library/Cura/2.6/machine_instances including those that were not successfully imported into Cura 2.7

Ghostkeeper, 2017-09-10:
Interesting tidbit from torque's log:

    2017-09-04 13:03:00,032 - DEBUG - cura.Settings.CuraContainerRegistry._convertContainerStack [372]: Converting ContainerStack Ultimaker 3 Extended to machine
    2017-09-04 13:03:00,035 - WARNING - UM.Logger.logException [76]: Exception: Exception in machine_stack upgrade with VersionUpgrade25to26.VersionUpgrade25to26: Traceback (most recent call last):
      File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/VersionUpgradeManager.py", line 370, in updateFilesData
      File "/Users/USER/Downloads/Cura.app/Contents/MacOS/plugins/plugins/VersionUpgrade/VersionUpgrade25to26/VersionUpgrade25to26.py", line 130, in upgradeMachineStack
        definition_container_id = parser["containers"]["6"]
      File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/configparser.py", line 1230, in __getitem__
    KeyError: '6'

hillct, 2017-10-22:
Having downloaded installed/updated to Cura 3.0, it appears this issue has not yet been resolved and failures to import profiles, extend now to those created in Cura 2.7, in addition to the previously mentioned 2.6

Ghostkeeper, 2017-10-31:
One other thing we could try is if you'd go to Help -> Show Configuration Folder... in Cura and .zip up that entire folder and send it to us (or at least the folder of your older version). That .zip file will contain all of your profiles and settings, so if there's something wrong with those profiles it'll be in there.

ianpaschal, 2018-03-05:
@Ellecross

hillct, 2018-04-28:
This issue has been ongoing for every subsequent release I've used. As of installing 3.2.1, I deleted all materials, machine and print profiles (from within cura - I probably should have deleted the application support/cura folder, in retrospect) which worked fine for 3.2.1. I skipped 3.3.0 and today updated to 3.3.1 which failed to start at all, reporting a problem with config data, prompting me to send a crash report (in which I referenced this bug report) and backup/delete/clear my config to try again. I archived the old config directory for provision here (see attached) and proceeded to submit the bug report, and let Cura do whatever recovery it was going to attempt, deleting the config data to whatever degree it was going to. SO far, it has not restarted, and it's not clear whether it was meant to do so or I was meant to restart it, after that operation. Visually, the cura config directory seems relatively untouched, other than the automatic generation of a zip file of the v3.3 subdirectory, which I imagine was uploaded automatically with the crash report. I've attached here, for the v3.3 directory archive that was generated, as well as my own archive of the parent directory, containing the directories for all previous Cura releases used, for reference, so you might be able to trace the origin of this issue more thoroughly.

This has become a sort of stream of consciousness, to the extent that I now see that I'm unable to attach the zip files of the config directories as they're both larger than the 10MB limit Github has in place. The v3.3 directory is 448MB in size and the entire (from parent directory) config directory tree is 902MB. Please advise as to how you prefer that I provide these files to you for review.

-- Colin

nallath, 2019-01-07:
Closing due to inactivity.

closed 2019-01-07 by nallath