Ultimaker/Cura #2364: Cura 2.6.1 on ubuntu 16.04  (not connected to a printer)

Elemino, 2017-09-01:
Hello, 

Cura is not recognizing my printrbot simple printer. I was wondering if any of you guys have some troubleshooting tips?

Thank you so much for any help! 

Cheers.

LipuFei, 2017-09-04:
Hi @Elemino, I suspect a permission problem (no permission to read/write to the device). Could you provide your log file? Thanks!
On LInux, the log file `cura.log` is in `~/.local/share/cura/`

Elemino, 2017-09-05:
Thanks a bunch! @LipuFei 

speed_support_infill="60" -s speed_equalize_flow_max="150" -e0 -l "0"
2017-09-01 13:52:31,474 - DEBUG - CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [474]: Slicing took 2.322845697402954 seconds
2017-09-01 13:52:31,475 - DEBUG - UM.Backend.Backend._logSocketState [176]: Socket state changed to Closing
2017-09-01 13:52:31,476 - DEBUG - UM.Backend.Backend._onSocketError [198]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2017-09-01 13:52:31,477 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Closed
2017-09-01 13:52:31,478 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [585]: Backend quit with return code 0. Resetting process and socket.
2017-09-01 13:52:39,637 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-09-01 14:47:05,721 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [379]: Closing the USB printer connection.
2017-09-01 14:47:25,758 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [311]: Attempting to connect to /dev/ttyACM0
2017-09-01 14:47:25,759 - INFO - USBPrinting.USBPrinterOutputDevice._connect [319]: Could not establish connection on /dev/ttyACM0: 'Failed to open serial port'. Device is not arduino based.
2017-09-01 14:47:25,760 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 115200
2017-09-01 14:47:25,761 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [125]: Could not find output device with id /dev/ttyACM0 to remove
2017-09-01 14:47:25,762 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
2017-09-01 14:47:25,762 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 250000
2017-09-01 14:47:25,763 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
2017-09-01 14:47:25,764 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 230400
2017-09-01 14:47:25,765 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
2017-09-01 14:47:25,765 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 57600
2017-09-01 14:47:25,766 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
2017-09-01 14:47:25,766 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 38400
2017-09-01 14:47:25,767 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
2017-09-01 14:47:25,768 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 19200
2017-09-01 14:47:25,769 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
2017-09-01 14:47:25,769 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 9600
2017-09-01 14:47:25,770 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
2017-09-01 14:47:25,771 - ERROR - USBPrinting.USBPrinterOutputDevice._connect [365]: Baud rate detection for /dev/ttyACM0 failed
2017-09-01 14:47:25,771 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [379]: Closing the USB printer connection.
2017-09-01 14:47:25,772 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [384]: PrinterConnection.close: cannot join current thread (expected)
2017-09-01 14:47:25,772 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [125]: Could not find output device with id /dev/ttyACM0 to remove
2017-09-01 14:47:30,764 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [379]: Closing the USB printer connection.
2017-09-01 14:47:35,777 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [311]: Attempting to connect to /dev/ttyACM0
2017-09-01 14:47:35,778 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [379]: Closing the USB printer connection.
2017-09-01 14:47:35,779 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [379]: Closing the USB printer connection.
2017-09-01 14:47:35,780 - INFO - USBPrinting.USBPrinterOutputDevice._connect [319]: Could not establish connection on /dev/ttyACM0: 'Failed to open serial port'. Device is not arduino based.
2017-09-01 14:47:35,780 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 115200
2017-09-01 14:47:35,781 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
2017-09-01 14:47:35,781 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 250000
2017-09-01 14:47:35,782 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
2017-09-01 14:47:35,783 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [327]: Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 230400
2017-09-01 14:47:35,783 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [333]: Could not open port /dev/ttyACM0
[... cut ...]
2017-09-01 14:48:14,124 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/platform.shader]...
2017-09-01 14:48:21,780 - DEBUG - UM.Qt.QtApplication.windowClosed [267]: Shutting down cura
2017-09-01 14:48:21,791 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [235]: Attempting to kill the engine process
2017-09-01 14:48:21,791 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [241]: Killing engine process
2017-09-01 14:48:21,792 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [244]: Engine process is killed. Received return code -15

Elemino, 2017-09-05:
Here the full log. https://gist.github.com/Elemino/6511398782b40249c9cc53bb2aa90ba6

fieldOfView, 2017-09-05:
Could you try Cura 2.7? The way printers are detected was changed in Cura 2.7, and it should be a little more permissive. What printer/controller board are you using, and what port/device is it on?

LipuFei, 2017-09-05:
Hi @Elemino, you can try Cura 2.7 as @fieldOfView has mentioned.

Or you could also check the permissions of `/dev/ttyACM0` by running the command below and see if you get a "permission denied" error.
```
cat /dev/ttyACM0
```

Elemino, 2017-09-06:
@LipuFei  Here're the contents of /dev 

/dev ls -la       
total 4
drwxr-xr-x  20 root root        4260 Sep  5 19:50 .
drwxr-xr-x  24 root root        4096 Aug 29 10:58 ..
crw-------   1 root root     10, 235 Sep  5 16:01 autofs
drwxr-xr-x   2 root root         300 Sep  5 16:01 block
drwxr-xr-x   2 root root          80 Sep  5 16:01 bsg
crw-------   1 root root     10, 234 Sep  5 16:01 btrfs-control
drwxr-xr-x   3 root root          60 Sep  5 16:01 bus
lrwxrwxrwx   1 root root           3 Sep  5 16:01 cdrom -> sr0
lrwxrwxrwx   1 root root           3 Sep  5 16:01 cdrw -> sr0
drwxr-xr-x   2 root root        4380 Sep  5 22:17 char
crw-------   1 root root      5,   1 Sep  5 16:01 console
lrwxrwxrwx   1 root root          11 Sep  5 16:01 core -> /proc/kcore
drwxr-xr-x   2 root root          60 Sep  5 16:01 cpu
crw-------   1 root root     10,  59 Sep  5 16:01 cpu_dma_latency
crw-------   1 root root     10, 203 Sep  5 16:01 cuse
drwxr-xr-x   5 root root         100 Sep  5 16:01 disk
drwxr-xr-x   2 root root         100 Sep  5 16:01 dri
lrwxrwxrwx   1 root root           3 Sep  5 16:01 dvd -> sr0
lrwxrwxrwx   1 root root           3 Sep  5 16:01 dvdrw -> sr0
crw-------   1 root root     10,  61 Sep  5 16:01 ecryptfs
crw-rw----   1 root video    29,   0 Sep  5 16:01 fb0
lrwxrwxrwx   1 root root          13 Sep  5 16:01 fd -> /proc/self/fd
crw-rw-rw-   1 root root      1,   7 Sep  5 16:01 full
crw-rw-rw-   1 root root     10, 229 Sep  5 16:01 fuse
crw-------   1 root root    248,   0 Sep  5 16:01 fw0
crw-------   1 root root    246,   0 Sep  5 19:50 hidraw0
crw-------   1 root root    246,   1 Sep  5 19:50 hidraw1
crw-------   1 root root    246,   2 Sep  5 19:50 hidraw2
crw-------   1 root root     10, 228 Sep  5 16:01 hpet
drwxr-xr-x   2 root root           0 Sep  5 16:01 hugepages
crw-------   1 root root     10, 183 Sep  5 16:01 hwrng
crw-------   1 root root     89,   0 Sep  5 16:01 i2c-0
crw-------   1 root root     89,   1 Sep  5 16:01 i2c-1
crw-------   1 root root     89,  10 Sep  5 16:01 i2c-10
crw-------   1 root root     89,  11 Sep  5 16:01 i2c-11
crw-------   1 root root     89,   2 Sep  5 16:01 i2c-2
crw-------   1 root root     89,   3 Sep  5 16:01 i2c-3
crw-------   1 root root     89,   4 Sep  5 16:01 i2c-4
crw-------   1 root root     89,   5 Sep  5 16:01 i2c-5
crw-------   1 root root     89,   6 Sep  5 16:01 i2c-6
crw-------   1 root root     89,   7 Sep  5 16:01 i2c-7
crw-------   1 root root     89,   8 Sep  5 16:01 i2c-8
crw-------   1 root root     89,   9 Sep  5 16:01 i2c-9
lrwxrwxrwx   1 root root          25 Sep  5 16:01 initctl -> /run/systemd/initctl/fifo
drwxr-xr-x   4 root root         500 Sep  5 19:50 input
crw-r--r--   1 root root      1,  11 Sep  5 16:01 kmsg
crw-rw----+  1 root root     10, 232 Sep  5 16:01 kvm
drwxr-xr-x   2 root root          60 Sep  5 16:01 lightnvm
lrwxrwxrwx   1 root root          28 Sep  5 16:01 log -> /run/systemd/journal/dev-log
brw-rw----   1 root disk      7,   0 Sep  5 16:01 loop0
brw-rw----   1 root disk      7,   1 Sep  5 16:01 loop1
brw-rw----   1 root disk      7,   2 Sep  5 16:01 loop2
brw-rw----   1 root disk      7,   3 Sep  5 16:01 loop3
brw-rw----   1 root disk      7,   4 Sep  5 16:01 loop4
brw-rw----   1 root disk      7,   5 Sep  5 16:01 loop5
brw-rw----   1 root disk      7,   6 Sep  5 16:01 loop6
brw-rw----   1 root disk      7,   7 Sep  5 16:01 loop7
crw-rw----   1 root disk     10, 237 Sep  5 16:01 loop-control
drwxr-xr-x   2 root root          60 Sep  5 16:01 mapper
crw-------   1 root root     10, 227 Sep  5 16:01 mcelog
crw-------   1 root root    245,   0 Sep  5 16:01 media0
crw-r-----   1 root kmem      1,   1 Sep  5 16:01 mem
crw-------   1 root root     10,  56 Sep  5 16:01 memory_bandwidth
drwxrwxrwt   2 root root          40 Sep  5 16:01 mqueue
drwxr-xr-x   2 root root          60 Sep  5 16:01 net
crw-------   1 root root     10,  58 Sep  5 16:01 network_latency
[... 128 lines cut ...]
crw-rw----   1 root tty       7,   1 Sep  5 16:01 vcs1
crw-rw----   1 root tty       7,   2 Sep  5 16:01 vcs2
crw-rw----   1 root tty       7,   3 Sep  5 16:01 vcs3
crw-rw----   1 root tty       7,   4 Sep  5 16:01 vcs4
crw-rw----   1 root tty       7,   5 Sep  5 16:01 vcs5
crw-rw----   1 root tty       7,   6 Sep  5 16:01 vcs6
crw-rw----   1 root tty       7, 128 Sep  5 16:01 vcsa
crw-rw----   1 root tty       7, 129 Sep  5 16:01 vcsa1
crw-rw----   1 root tty       7, 130 Sep  5 16:01 vcsa2
crw-rw----   1 root tty       7, 131 Sep  5 16:01 vcsa3
crw-rw----   1 root tty       7, 132 Sep  5 16:01 vcsa4
crw-rw----   1 root tty       7, 133 Sep  5 16:01 vcsa5
crw-rw----   1 root tty       7, 134 Sep  5 16:01 vcsa6
drwxr-xr-x   2 root root          60 Sep  5 16:01 vfio
crw-------   1 root root     10,  63 Sep  5 16:01 vga_arbiter
crw-------   1 root root     10, 137 Sep  5 16:01 vhci
crw-------   1 root root     10, 238 Sep  5 16:01 vhost-net
crw-rw----+  1 root video    81,   0 Sep  5 16:01 video0
crw-rw-rw-   1 root root      1,   5 Sep  5 16:01 zero

Is there an extra permissions layer that locks the files? 

Cheers! 

@fieldOfView Will update to 2.7 as needed in case it is not a permissions issue. Thanks!

Elemino, 2017-09-06:
****

Ghostkeeper, 2017-09-11:
We test the connection by sending the command `M105` to the printer, which gets the hot-end temperature. Maybe you can load up Pronterface or Arduino IDE and see if the printer responds to that if you send it from there?

Elemino, 2017-09-15:
I am struggling with installing Pronterface. Currently at 

./pronterface.py               
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
WARNING:root:Could not setup DBus for sleep inhibition: No module named dbus
WARNING:root:psutil unavailable, could not import power utils:No module named psutil
WARNING:root:RPC server bound on non-default port 7979
[ERROR] 3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "/home/user/Downloads/Printrun-master/printrun/gui/viz.py", line 70, in __init__
    import printrun.gcview
  File "/home/user/Downloads/Printrun-master/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/home/user/Downloads/Printrun-master/printrun/gl/panel.py", line 21, in <module>
    import numpy
ImportError: No module named numpy

I am running Cura 2.7.

Elemino, 2017-09-16:
Nevermind. Got it running by cloning the repo. Will update asap.

Elemino, 2017-09-16:
It is indeed a permission issue. When I run protnerface.py as admin, the printer responds. 

I just wonder if there's a command that I can use to release the port or whatever is blocking.

Thanks for your help! Cheers!

closed 2017-09-16 by Elemino

Elemino, 2017-09-19:
I am not sure why but my experience with Cura on Linux is way better. I keep getting much more accurate prints on the first go.

Ghostkeeper, 2017-09-25:
I know why: Because half of our developers are on Linux ;)

Elemino, 2017-09-29:
Makes sense! 

Correction: I think that what I mean is that I find myself on average tweaking settings less while getting overall better print results in terms of completion rather than accuracy. Thanks for your help guys! Keep up the great work!