Ultimaker/Cura #2365: Cura 2.7 unable to select nested materials

stephanpark, 2017-09-01:
Unable to select materials that require drill down.  Only one level opens rarely allowing selection and actual material to be used for print.  Only able to select "Generic" materials because they are not nested.

LipuFei, 2017-09-04:
Hi @stephanpark, by "unable to select", do you mean that the options are there but when you click, it doesn't react, or there is no multi-level options for you to select at all?
And could you provide your log file as well? Thanks.

stephanpark, 2017-09-05:
The UI element pops up but hot-zone is missing thus not triggering sub item flyout.  It instead triggers whatever is beneath such as "more-info" info bubbles from anything behind it.  Log file location unknown, no errors triggered to have seen an error dialog, where is it?

LipuFei, 2017-09-05:
The log file is here: `C:\Users\<your-username>\AppData\Roaming\cura\2.7\cura.log` on Windows.

nallath, 2017-09-05:
No machines that we currently support have settings that depend on "specific" materials. If you select red ultimaker PLA, it will use the profiles as provided by the generic material.

As for not being able to select them; i fear that there is little we can do about this. The generation of those popups is done by our GUI library, Qt.

Ghostkeeper, 2017-09-11:
I'm not entirely sure what you mean, actually. What UI element is this issue talking about? The specific-colour materials in the list of materials? What do you mean with "sub item flyout"?

ianpaschal, 2018-03-02:
Needs info but never got it.

closed 2018-03-02 by ianpaschal