Ultimaker/Cura #2366: Extensions, Post Processing, Modify G-code: Where are the scripts located (on Windows)?

b-morgan, 2017-09-01:
On previous versions of Cura (21.04 Lulzbot Edition), there was a button on the script selection screen to open the folder. In that folder were the source code for the existing scripts, and one could place new (or modified) scripts in that folder which would then be available for use.

In Cura 2.7 beta, it isn't clear where the folder is located or how one would add a script to the "Extensions, Post Processing, Modify G-code" menu.

If the answer is OS dependent, then please include the answer for Windows 10.

nallath, 2017-09-04:
They are located in the scripts folder, which in turn can be found in the plugins/postprocessingplugin folder (which is in your program files)

closed 2018-01-17 by ChrisTerBeke

fieldOfView, 2018-01-17:
For the benefit of those stumbling across this: Since Cura 3.0, there is also a "scripts" folder in the configuration folder (Help -> Show configuration folder)