Ultimaker/Cura #2367: Cura 2.6 and 2.7 fail to load

3DKernow, 2017-09-02:
For the last 2-3 months, I have sporadically tried to load wither 2.6 or (more recently) 2.7. I follow instructions and close other programmes and uninstall 2.5. Consistently, the following then happens: (i) the .exe file loads fine (ii) Norton confirms that the file is safe (iii) the installation process starts but around half way on the moving ribbon, it cuts out and tells me that installations is complete (iv) it then seems to try to boot the programme up, and it shows up in Task Manager as being active for about 40 seconds, at which point all Cura activity seems to stop. 

When I reinstall 2.5, I have no problems. I am running on an up-to-date Windows, with Norton 360 premier as my anti virus programme.

Two observations. (i) in settings - the relevent version of Cura shows up, but without any indication of file size (ii) The initial breakdown of the installation process seems to be just after it's dealing with a file with a title like "norestart"

nallath, 2017-09-04:
Could you share the log files?

3DKernow, 2017-09-04:
How do I record those?

nallath, 2017-09-04:
https://github.com/Ultimaker/Cura#logging-issues

3DKernow, 2017-09-04:
Jaime: the relevant log files. Let me know if you need anything else. Louis

 

 

2017-09-01 13:38:21,152 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger

2017-09-01 13:38:21,152 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger

2017-09-01 13:38:21,219 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeProfileReader

2017-09-01 13:38:21,222 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeProfileReader

2017-09-01 13:38:21,296 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeReader

2017-09-01 13:38:21,298 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeReader

2017-09-01 13:38:21,371 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeWriter

2017-09-01 13:38:21,373 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin GCodeWriter

2017-09-01 13:38:21,497 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ImageReader

2017-09-01 13:38:21,499 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ImageReader

2017-09-01 13:38:21,638 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LayerView

2017-09-01 13:38:21,639 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LayerView

2017-09-01 13:38:21,698 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LegacyProfileReader

2017-09-01 13:38:21,699 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LegacyProfileReader

2017-09-01 13:38:21,778 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice

2017-09-01 13:38:21,782 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice

2017-09-01 13:38:21,846 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MachineSettingsAction

2017-09-01 13:38:21,848 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MachineSettingsAction

2017-09-01 13:38:22,312 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OctoPrintPlugin

2017-09-01 13:38:22,313 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OctoPrintPlugin

2017-09-01 13:38:22,488 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PerObjectSettingsTool

2017-09-01 13:38:22,491 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PerObjectSettingsTool

2017-09-01 13:38:22,569 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PluginBrowser

2017-09-01 13:38:22,571 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PluginBrowser

2017-09-01 13:38:22,680 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PostProcessingPlugin

2017-09-01 13:38:22,681 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin PostProcessingPlugin

2017-09-01 13:38:22,876 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RemovableDriveOutputDevice

2017-09-01 13:38:22,881 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RemovableDriveOutputDevice

2017-09-01 13:38:23,294 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SliceInfoPlugin

2017-09-01 13:38:23,295 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SliceInfoPlugin

2017-09-01 13:38:23,370 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SolidView

2017-09-01 13:38:23,371 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SolidView

[... 1108 lines cut ...]
2017-09-01 13:38:30,871 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_extruder_right_normal+new+tweaks.inst.cfg to version 2000002.

2017-09-01 13:38:30,888 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_extruder_right_normal+quality+%232.inst.cfg to version 2000002.

2017-09-01 13:38:30,905 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_extruder_right_normal+scaff+tower.inst.cfg to version 2000002.

2017-09-01 13:38:30,921 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_extruder_right_simple+scaffolding.inst.cfg to version 2000002.

2017-09-01 13:38:30,939 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_fast+print+max+support.inst.cfg to version 2000002.

2017-09-01 13:38:30,957 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_normal+new+tweaks.inst.cfg to version 2000002.

2017-09-01 13:38:30,973 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_normal+quality+%232.inst.cfg to version 2000002.

2017-09-01 13:38:30,990 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_normal+scaff+tower.inst.cfg to version 2000002.

2017-09-01 13:38:31,005 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_simple+scaffolding.inst.cfg to version 2000002.

2017-09-01 13:38:31,027 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening

2017-09-01 13:38:31,096 - DEBUG - UM.Backend.Backend._onSocketError [200]: Socket was unable to bind to port, increasing port number to 49675

2017-09-01 13:38:31,102 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.

2017-09-01 13

nallath, 2017-09-04:
You need to clear out your local files; It seems to have one corrupted file that is preventing Cura from starting.

3DKernow, 2017-09-04:
Jaimie

 

Still having the same problem after purging everything in the local/cura directory, and after purging anything with “temp” in the file name in the rest of this local directory.

Are you actually asking me to purge EVERYTHING in this C:\Users\louis_000\AppData\Local directory. That would scare me since so many other programmes could be involved.

 

Are there any clues as to where this corrupted file could be?

 

The following screenshots show what I’m looking at. The Cura directory is empty.

 

Louis

nallath, 2017-09-04:
I'm only asking you to purge the local directory for Cura. 

As for the screenshots, they aren't added. Could you add them via Github (and not via email?)

3DKernow, 2017-09-04:
Jaimie

If you can see the attached screenshots, they merely show you what is in my "local" directory. Just to confirm, I can still keep on loading and uninstalling Cura 2.5. However, cura 2.6 and 2.7 still won't install. As far as I can see, I've purged all files in the local/cura directory.

[image]

[image]

fieldOfView, 2017-09-04:
There's also a cura folder in AppData/Roaming.

3DKernow, 2017-09-04:
Thanks. That led to 2.7 installing, but it seems to be a little incomplete. It has the UM3 down as the only supported printer, but only offers me a "fine" print option, and won't allow me to assign the second print head to PVA.

I can't work on this any more today. I probably won't be able to get back to this until tomorrow evening.

3DKernow, 2017-09-05:
All now seems well. I'd forgotten that PVA needs its own BB printhead to be set. I've got my UM2 also supported. Many thanks.

Ghostkeeper, 2017-09-11:
It's still a mystery how that file got corrupted.

The actual error says that a machine instance didn't have any "containers" category. It should always have a "containers" category. Maybe the file write to the temporary file failed, but then it continued and renamed the file to its permanent location anyway?

ChrisTerBeke, 2017-09-27:
Closing until issue resurfaces?

closed 2017-09-27 by ChrisTerBeke

3DKernow, 2017-09-27:
Your team successfully identified the rogue file which was causing this download to fail. All now OK

 

Thanks