Ultimaker/Cura #2370: Crash opening STL - Bug Report v2.7beta

peepsnet, 2017-09-02:
Wanted to open these STLs:
https://www.thingiverse.com/thing:2153075

Version: 2.7.0-BETA
Platform: Windows-8.1-6.3.9600
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 707, in event
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 261, in event
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Event.py", line 189, in call
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 131, in calculateBoundingBoxMesh
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 596, in getBoundingBox
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 654, in _calculateAABB
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 180, in getExtents
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 255, in getConvexHullVertices
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 247, in getConvexHull
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 239, in _computeConvexHull
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 386, in approximateConvexHull
  File "scipy\spatial\qhull.pyx", line 2385, in scipy.spatial.qhull.ConvexHull.__init__ (scipy\spatial\qhull.c:24141)
  File "scipy\spatial\qhull.pyx", line 434, in scipy.spatial.qhull._Qhull.__init__ (scipy\spatial\qhull.c:5799)
scipy.spatial.qhull.QhullError: QH6013 qhull input error: input is less than 3-dimensional since it has the same x coordinate

While executing:  | qhull i Qt
Options selected for Qhull 2015.2.r 2016/01/18:
  run-id 1486384747  incidence  Qtriangulate  _pre-merge  _zero-centrum
  _max-width  0  Error-roundoff  0  _one-merge  0  _near-inside  0
  Visible-distance  0  U-coplanar-distance  0  Width-outside  0  _wide-facet  0

-------------------------
Thread 0x00000bcc (most recent call first):
  File "C:\Program Files\Cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002d24 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00003228 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000030f0 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00002ac0 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap
[... 85 lines cut ...]
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000016cc (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x00001054 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 676 in run
  File "D:/2.7/build/inst/bin/cura_app.py", line 73 in <module>
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

Thisismydigitalself, 2017-09-02:
2.7 was released.  
Go for it: https://ultimaker.com/en/blog/50847-introducing-cura-27

peepsnet, 2017-09-03:
Did not crash in 2.7

closed 2017-09-03 by peepsnet