Ultimaker/Cura #2371: Z-hop slow travel with Cura 2.7

Miranio, 2017-09-02:
Since the update to 2.7 I noticed a weird behavior of my printer. Sometimes after the z-hop is done, it will travel verrrrry slowly to the start of the next layer. I opened the gcode and looked at those.

When it travel normally the end layer gcode will look like this : 
```
G1 X98.009 Y89.675 E166.1461
G0 F7200 X98.208 Y89.663
G1 F1500 E162.6461
G1 F300 Z4.1
G0 F7200 X44.765 Y52.262 Z4.1
G0 X44.115 Y51.584
;TIME_ELAPSED:7316.893784
;LAYER:19
```

But when it travel super slowly it will be like this :
```
G1 F4200 X102.47 Y86.467 E91.92936
G0 F7200 X102.351 Y86.533
G1 F1500 E88.42936
G1 F300 Z3.9
G0 F7200 X102.374 Y86.644 Z3.9
G0 X160.196 Y52.308
G1 F300 Z3.9
G0 X44.47 Y51.278
;TIME_ELAPSED:7273.878708
;LAYER:18
```

Note the last G0 command is missing a FeedRate I suspect the printer revert to the last feedrate of 300.

If I disable z-hop it work correctly but ... z-hop is kind of usefull.

smartavionics, 2017-09-02:
Hi, actually, that G1 F300 Z18.7 that appears to be setting the slow feed rate is completely spurious because Z is already 18.7. So how did that come to get there?

Miranio, 2017-09-02:
Don't ask me ... I don't make Cura software. I noticed the bug and have a basic, understanding of gcode but that's the extend of it.

Btw there is no plugin, it's a vanilla Cura I'm using.

smartavionics, 2017-09-02:
Do you have retract at layer change enabled? If so, please try with that disabled and see if it makes any difference.

smartavionics, 2017-09-02:
Sorry, I didn't really mean for you to answer that question, I was talking to myself!

Miranio, 2017-09-02:
Retract at layer change is already off.

There is my current profile. Only difference is the z-hop that is enable when it do the slooooooow travel

definition = fdmprinter

[metadata]
type = quality_changes
quality_type = draft
setting_version = 2

[values]
acceleration_enabled = False
acceleration_wall = 2000
adhesion_type = skirt
carve_multiple_volumes = True
coasting_enable = False
cool_fan_speed = 75
fill_perimeter_gaps = nowhere
gradual_infill_step_height = 5
gradual_infill_steps = 1
infill_pattern = cubic
infill_sparse_density = 0
ironing_enabled = False
ironing_flow = 5
jerk_enabled = False
jerk_wall = 15
material_bed_temperature = 45
material_bed_temperature_layer_0 = 55
material_flow = 96
material_print_temperature = 205
meshfix_extensive_stitching = True
retraction_amount = 3.5
retraction_extra_prime_amount = 0.08
retraction_hop = 0.2
retraction_hop_enabled = False
retraction_min_travel = 0.8
skirt_line_count = 2
speed_layer_0 = 30
speed_print = 60
speed_topbottom = 70
speed_wall = 60
support_infill_rate = 10
support_roof_enable = False
support_use_towers = False

smartavionics, 2017-09-02:
OK, there was a bug in that the feed rate used for the z-hop was then used for the following X/Y move. I have created a fix for that which the Cura developers will think about. I still don't understand why it generated that second G1 F300 Z18.7, perhaps they can shed some light on that. I will think about it some more tomorrow but it's bedtime here!

smartavionics, 2017-09-02:
One last thought, can you try enabling  z hop when retracted and see if that makes any difference. Thanks.

smartavionics, 2017-09-03:
Ignore that last request, I've just noticed your comment above where you mention that the slow travel happens when z-hop is enabled.

smartavionics, 2017-09-03:
Cura devs, I think the second G1 F300 Z18.7 before the layer change is there because there are two separate travels (each z-hopping) with no extrusion between them.

smartavionics, 2017-09-03:
The completely spurious travel to Z18.9 should be fixed with https://github.com/Ultimaker/CuraEngine/pull/600.

cdeerinck, 2017-09-08:
I saw this issue in the 2.6 beta, and again in 2.7.  It usually had to do with just the last move between layers, but on a print today, I noticed it in some other moves.

Here is an example of a mid-Layer instance of the bug.  The F300 (300mm/minute) equates to 5mm/s and that speed is set nowhere in any of my speeds.

G1 F2700 E360.74461
**G1 F300 Z8.2**
G0 F7200 X-9.995 Y11.191 Z8.2
G0 X7.233 Y-12.685
;TIME_ELAPSED:678.441144
;LAYER:36
M106 S74.6
G0 X7.233 Y-12.685 Z8.4
;TYPE:WALL-INNER
**G1 F300 Z7.4**
G1 F2700 E370.74461
G1 F1800 X7.13 Y-12.783 E370.7494
G1 X7.084 Y-12.867 E370.75262
G1 X7.058 Y-12.961 E370.7559
G1 X7.055 Y-13.041 E370.75859

Here is a mid-Layer instance, again using an F300
G1 F2700 E360.99735
**G1 F300 Z8.4**
G0 F7200 X7.45 Y-13.01 Z8.4
G0 X11.822 Y-9.145
;TYPE:WALL-INNER
**G1 F300 Z7.4**
G1 F2700 E370.99735

FYI, my Inner Wall Speed is set to 30.0 mm/s, which equates to F2700.

I have an object that has 15 different vertical supports, and strangely on some layers the move is slow between each one, on some it is never slow, and on some it is a mix of the two.

smartavionics, 2017-09-08:
HI. I think that is a related issue but not actually the same bug as discussed above.
Around that time, one of the Cura devs added the feedrate (set by machine_max_feedrate_z) to the z-hop which you are seeing above. I had a problem with that causing very slow moves so I added this to my machine definitions to set the feedrate to what I would like it to be:
```
        "machine_max_feedrate_z": {
            "default_value": 300
        }
```

The default value of 5 mm/S maybe OK for an Ultimaker machine but for a delta it's ludicrous.

smartavionics, 2017-09-08:
If you give the Maximum Z Speed setting a value other than 0 then that will be used instead of machine_max_feedrate_z  and an M203 will get generated to set the max z feedrate used for the z-hops. Unfortunately, there's a problem with that for some printer firmwares. If you are using RepRap flavour gcode the M203 specifies the speed in mm/min but for all other gcode flavours it specifies the speed in mm/s because that is what Marlin expects. So unless you are using RepRap or a printer firmware that also expects the M203 speed to be in mm/s then it will be set to a value that is 60 times lower than what you wanted.
As far as I know, the only way to fix the slow hop bug is to set the feedrate (in mm/s) in the machine definition file (machine_max_feedrate_z) and then ensure that the "Maximum Z Speed" setting is set to 0. Confusing, or what?

Ghostkeeper, 2017-09-15:
> The default value of 5 mm/S maybe OK for an Ultimaker machine but for a delta it's ludicrous.

It's okay for XYZ printers basically because they typically don't have a whole lot of power behind their Z axis. I think that a delta printer should just override the `machine_max_feedrate_z` because they can move in the Z direction much faster.

smartavionics, 2017-09-15:
Indeed, for my printers, I have set machine_max_feedrate_z to 300 which they can handle.

BTW - I discovered my fix for this bug was flawed and have submitted a new engine PR to fix the fix https://github.com/Ultimaker/CuraEngine/pull/608

pouncingiguana, 2017-12-05:
This bug is still present in Cura 3.1.0! Is this going to get fixed any time soon?

smartavionics, 2017-12-05:
Hi @pouncingiguana , could you please post some gcode that exhibits this problem so I can see what's happening - thanks.

pouncingiguana, 2017-12-05:
Just sliced this on Cura 3.1.0 running on Linux. It's not respecting the max z speed and putting it at F300:
...
G1 F2700 E22.95254
G1 F300 Z1.2
G0 F9000 X6.94 Y-14.527 Z1.2
;TYPE:WALL-INNER
G1 F300 Z0.2
G1 F2700 E25.99286
...

smartavionics, 2017-12-05:
Sorry, could you please post a link to the entire gcode file or upload a zip of it?

smartavionics, 2017-12-05:
BTW, where have you set the higher z speed?

Miranio, 2017-12-05:
This bug wasn't about the z speed but about the x-y travel speed after a Z hop. The later issue is actually fixed.

As to specifying F300 when zspeed is set to 0 when it's supposed to use the maximum speed of the machine, I noted this bug too. Simply specifying a value like 32mm/sec for z speed and it work great. Though I'm still on 3.0.3.

smartavionics, 2017-12-05:
Please make the gcode file available...

smartavionics, 2017-12-05:
Sorry @Miranio, I was replying to @pouncingiguana. The original issue of the slow x-y speed should be fixed in 3.1.

Miranio, 2017-12-05:
@smartavionics The original issue WAS fixed in Cura 3.x ( thanks to you and you were credited in the release note !! ) That is why I responded. I'm unsure if this would more be a new bug than a reopening of this one.

smartavionics, 2017-12-05:
I can't remember when it happened, not too many releases ago, a regression was introduced whereby the z speed used for z-hops was too slow. I thought we had fixed that so I am waiting for @pouncingiguana  to provide more info to see what is happening.

pouncingiguana, 2017-12-06:
Apologies, it appears that dragging a profile from an older version of Cura forward on Linux breaks a few things. Including this. I threw out everything from ~/.local/share/cura and ~/.config/cura and tried it again and it's working properly. Sorry for the misinformation.

smartavionics, 2017-12-06:
Hi @pouncingiguana , no problem, I'm glad it's working OK for you. Cheers, Mark

closed 2017-12-19 by Ghostkeeper

pouncingiguana, 2017-12-19:
Apologies - I forgot I was still fighting this issue until I saw the closed notice. I tracked down the problem to a bug where it's putting in an unwanted (and wrong) M203 command. Whatever value I set for Maximum Z Speed, gets put in the GCode as an M203, but it's setting mm/min instead of mm/sec, so it's 60 times too slow. This is what's causing my issue with the slow Z travles. It also add another M203 to the end of the gcode.

Beggining of Gcode:
;FLAVOR:Repetier
;TIME:75889
;Generated with Cura_SteamEngine 3.1.0
M190 S60
M104 S210
M109 S210
M82 ; absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
;LAYER_COUNT:360
;LAYER:0
M107
M203 Z500          <----------------------------------
G0 F9000 X45.26 Y16.487 Z0.2
;TYPE:SKIRT
G1 F1500 X44.997 Y43.112 E1.10699
G1 X44.951 Y43.693 E1.13122
G1 X44.87 Y44.1 E1.14848
G1 X44.319 Y46.305 E1.24297

End Gcode:
;TIME_ELAPSED:75889.687061
G1 F2700 E3545.45711
M203 Z5        <----------------------------------
M107
M104 S0
M140 S0
G92 E1
G1 E-1 F300
G28
M84
M82 ; absolute extrusion mode
M104 S0

smartavionics, 2017-12-19:
Hi. OK, so the problem here is that Cura loves doing things the Marlin way and when that conflicts with what other gcode firmwares expect then an exception has to be added. Marlin expects the speed in M203 to be in mm/s but other firmwares expect it in mm/m. The Cura code already know that RepRap firmware is like that and we can easily make it do the same for Repetier flavour. I will put together a PR to achieve that.

Ghostkeeper, 2018-01-04:
The [documentation](http://reprap.org/wiki/G-code#M203:_Set_maximum_feedrate) says that this should result in some temperature monitor being set?

smartavionics, 2018-01-04:
Yes, that's what the doc says, doesn't it. Perhaps they have changed the firmware since that was written. @pouncingiguana, are you sure that M203 really does set the speed rather than diddle with a temperature monitor?

smartavionics, 2018-01-04:
Hi @pouncingiguana, bit of a dumb question but are you 100% sure that your printer is actually running Repetier firmware? I ask because it does look like M203 does something completely different in Repetier firmware so it doesn't make sense that the m203 is affecting the feedrate on your machine unless it's running something other than Repetier firmware.

pouncingiguana, 2018-01-04:
The link you posted shows that M203 is "Set Maximum Feedrate", it's setting the fastest the printer can move on the given axis. I've also confirmed this by removing that line entirely and by changing the number to the correct number and watching the speed increase. It sets the maximum speed (For the Z axis in this case) and it's setting it in mm/sec instead of mm/minute.

smartavionics, 2018-01-04:
Yes, it does that for non-Repetier firmwares but it says for Repetier that it does something different. Please confirm that your firmware really is Repetier (and a version number if you have it) - thanks

pouncingiguana, 2018-01-04:
It looks like my version of Repetier (0.92) has been customized a lot - it's a different fork. Stock Repetier does use M203 for temperature frequency monitoring. I'd imagine in that case the code should just be removed when Repetier is selected so it's not messing with it unintentionally.

Ghostkeeper, 2018-01-22:
I'd like to have a confirmation first that it messes with stock Repetier as well.