Ultimaker/Cura #2372: Cura no Disk error

1ngwpower, 2017-09-02:
Error "Cura.exe - No Disk   There is no disk in the drive." Been using Cura for a year, love it. This morning opened Cura 2.6.2 and got this error, can not stop it, won't open. Running MS10. Removed from drive and reinstalled from this site, still getting error.
Uploaded Cura Version: 2.7 64. Started the same issue, now cannot restart W10 nor reboot from menu. Have to throw the power switch.
[image]
Did not create a photo of the error in 2.7.

ChrisTerBeke, 2017-09-04:
Hi! Could it be that your OS is running from another disk than D? Just to be clear, prior to 2.6.2 this didn't happen?

nallath, 2017-09-04:
There probably is a SD card drive marked as the D drive. If you rename that one to anything but D it shouldn't fail anymore. This is obviously a work around, which we expect to fix next release.

MFHFozzy, 2017-09-04:
You can also disable the sd card slot to work around it.

rjm01, 2017-09-05:
Thank You all who made suggestions. I uninstalled all of Cura from deep dark recess's of my C. Reloaded 2.7 and as long as I don't leave the Micro SD reader in a USB port the program runs as it should!

Ghostkeeper, 2017-09-11:
Duplicate of #1602.

closed 2017-09-11 by Ghostkeeper

idiophonics, 2017-09-30:
Im still getting the error with Cura 2.7.  I had a NAS drive connected using D:  and cura worked fine the first time.  Then I noticed my NAS D: drive had a red X on it and I could not connect to it.  So I re-mapped the network drive to a new letter and notice (even after restarting) that D: is not an option as a drive letter to map to.  I have also tried installing a USB flash drive and remapping the drive letter using disk manager and again the letter D is not available to map. I have uninstall cura and as much of it as I can find on the C: drive then reinstalled and still the same problem.

nallath, 2017-10-02:
Yeah, we had a boo-boo with our buildserver, which caused the issue again but because of a different reason. The 3.0 release (which will be out soon) should finally resolve this.

Ghostkeeper, 2017-10-11:
"Resolve" this by assuming that you're not using the X: drive...

RGN01, 2017-10-19:
Nallath, unfortunately v3.0 did not fix this issue. 

I've just installed 3.0 from 2.7. I get that dreaded 'no disk in drive X:" message. After lots of investigation I discovered that it is trying to access a drive allocated to a multi-memory card reader. The particular slot uses a card format that I haven't used for about a decade so I'm pretty sure that I've never used it with any prior version of Cura.

If I disconnect the card reader then it works OK but I need it for the other cards.

Can this not please default to one of the fixed drives, or at least check that the drive is accessible before committing to use it?

Ghostkeeper, 2017-10-30:
This is a duplicate, please try to keep this discussion in #1602 as much as possible to keep the information together.

idiophonics, 2017-10-31:
Sorry, that discussion #1602 was closed and did not allow me to add to the string of comments/posts so I had to post my question as a new discussion.

Kevin Hartman