Ultimaker/Cura #2374: Send to Octoprint startup instruction option selection

necro-nemesis, 2017-09-03:
Sending directly to Octoprint forces bed to be verified at pre-heat temperature before printing. This is already handled appropriately in Gcode startup script and causes a delay. This option should be able to be selected off. 

Example of where this is an issue:
-Gcode instructs bed pre-heat without wait using M140.
-Pre-heat time is used to run auto-level routine. 
-With Cura sending instruction to set and wait for the bed temp there is now an undesirable delay added where auto-level routine could have commenced without this redundant pre-heat instruction.

fieldOfView, 2017-09-03:
You can edit your Cura start gcode in Machine Settings

necro-nemesis, 2017-09-04:
Yes and I have custom GCode in my settings but in addition to the custom GCode Cura is pre-heating the bed from the program itself. Try sending a print to Octoprint and you will see the pre-heat window come up on the right hand side of your screen. It is insisting on seeing the bed at a pre-determined temperature before running the startup GCode. I suggest a selection box to deactivate this feature.

fieldOfView, 2017-09-04:
As far as I understand the issue, you should either remove the preheating gcode from the start gcode snippet in Cura, or in OctoPrint.

Note that if your start gcode does not contain the string ```{material_print_temperature}```, Cura will add it automatically. If you place eg ```G4 P0; {material_print_temperature}``` in your start gcode, Cura will not do this.

necro-nemesis, 2017-09-06:
I'll have to double check the string name in the start code string but as far as I know to use the values set in the gui of Cura the start code must reside in Cura as it isn't passed or recognized in Octoprint as the variable string name Octoprint uses. Either the string is named differently or it isn't interpreted correctly by Octoprint if memory serves but it's been awhile since I attempted this and there have been multiple revisions of both pieces of software. As it stands Cura has all my GCode as it presently assigns the the bed and nozzle temp values set within Cura's gui correctly. Now if the string name is critical to determining whether to add this pre-heat in I will have to check the string is named in the manner you stated. In essences what I understand then is if my GCode contains the specified form of preheat code, Cura sees this and omits it's own insertion of the code which is now to heat the bed and wait right at the onset of running the startup of which the wait aspect is what I don't want to occur.

closed 2018-03-02 by ChrisTerBeke