Ultimaker/Cura #2376: Proposed bed adhesion setting: connect isolated features on first layer

ShakeThat, 2017-09-03:
Small isolated features on the first layer, such as screw holes, often fail to adhere or to print perfectly. This could be ameliorated by connecting such features with automatically-generated "causeways" (either only in areas destined to become bottom skin, or, more aggressively, passing through unprinted areas and thus needing to be trimmed afterwards). Where possible, these connections could be made with wall or bottom pattern lines that would otherwise have been printed later anyway.

This idea came to mind after turning on "outer before inner walls" eliminated a bunch of problematical islands in the first layer of a print that were causing it to fail miserably. At a minimum, avoid printing tiny islands on the bottom layer if they are destined to be connected by the time you're done!

smartavionics, 2017-09-03:
Yes, printing the outer walls first can help if the inner walls are just little blobs.
BTW I have just contributed another setting idea to Cura that I find very useful for improving adhesion. It allows you specify the speed for first layer walls independently. So, for example, I have been printing some PETG. The walls were printed using 15 or 20 mm/S and the skin was then printed at anything up to 50 mm/S and the adhesion was good. I'm sure that a lot of people would find that a useful setting to have.

ShakeThat, 2017-09-03:
That sounds very useful. No more dog-slow bottom skins just so we can get good walls.

Ghostkeeper, 2017-09-10:
How about a way to increase the number of walls for just the first layer?

Ghostkeeper, 2017-09-10:
Hmm, how would this work exactly? Connecting all outer walls with each other using straight strips of lines? Would it connect all-to-all, or sort of like a MST? And how would we handle strips that intersect with each other?

I think I like the initial layer skin pattern better as concentric.

ShakeThat, 2017-09-10:
Some of the techniques that I use to address aspects of the problem "manually" are: (1) change the model to connect isolated walls on the bottom layer, such as screwholes, using a trench from the outer perimeter, one layer high by 0.1mm wide, so that a continuous perimeter is formed; (2) change the model to connect small external islands using a causeway, one layer high by about 2 lines wide; (3) create a cutting mesh one layer high encompassing the whole model and set it to "outer before inner walls"; (4) add a single solid layer to the bottom of the model, and when the print is done, drill and cut the necessary holes. I want to obviate these techniques.

The goal is to lay down the first set of walls as a continuous bead of plastic. I think a single connection to any isolated wall is sufficient (fully or multiply connected graph not necessary), and where it can be done with a line that would have been laid down later as part of the skin, so much the better (though that might be too ambitious).