Ultimaker/Cura #2377: non stop reading of drive.

nighshocker, 2017-09-04:
Hello, 
I have version 2.6 and when it is loaded it non stops reading my A drive. Why does it do that?
I can not use it like that. May you please fix this problem.

Thanks.

ChrisTerBeke, 2017-09-04:
Hi! Can you tell us a bit more about your system? Which OS and version are you running? And is your A drive in use? If so, what type of disk is inserted?

p.s. for next time: we have a template for creating issues that'll give us the information we need to start processing an issue, please find it [here](https://github.com/Ultimaker/Uranium/wiki/Bug-Reporting-Template). Thanks! :)

LipuFei, 2017-09-04:
@nighshocker Could you provide your log file? It's here: `C:\Users\<your-username>\AppData\Roaming\cura\2.7.0\cura.log`

nallath, 2017-09-04:
This is fixed for 2.7

closed 2017-09-04 by nallath