Ultimaker/Cura #2378: Splash screen low resolution on retina displays

ChrisTerBeke, 2017-09-04:
When running Cura on a retina MacBook Pro, the splash screen looks very pixelated. Probably higher resolution images needed for screens with higher DPI?

Application Version: 2.7 (but happened before as well)
Platform: MacOS
Display Driver: Integrated Intel graphics

Steps to Reproduce:
Start Cura -> watch splash screen

Actual Results:
Pixelated splash screen

Expected results:
A nice high-res splash screen (as it's the first thing you see!)

Ghostkeeper, 2017-09-10:
A redesign was planned already for the 3.0 release.

ChrisTerBeke, 2017-09-10:
Indeed, just to make everyone aware that we need high-res assets :)

ChrisTerBeke, 2017-10-18:
@riodoro1 indeed, it turned out to be quite complicated as we don't have access to all QT features yet in the splash screen phase. We're looking for alternative solutions but it will take some time.

closed 2018-11-26 by Appesteijn