Ultimaker/Cura #2379: After Online Print, Cura 2.7 takes all free memory

konvoj, 2017-09-04:
During online printing Cura 2.7 is taking more and more memory, until it takes all free RAM making the computer unresponsive. Cura must be manually shut down in Task manager to get rid of it. See screenshot:

[image]

This happened to me already several times, so it is not one time issue.

fieldOfView, 2017-09-04:
Just in case it is the camera stream that's causing this:
https://github.com/Ultimaker/Cura/pull/2380

fieldOfView, 2017-09-04:
Probably a duplicate of https://github.com/Ultimaker/Cura/issues/2345

konvoj, 2017-09-04:
Looks indeed like a duplicate of the mentioned issues.

closed 2017-09-10 by Ghostkeeper

J0015, 2017-09-13:
Same problem here.  First I knew about it was when SolidWorks reported "Your system is running low on memory.  Executing this command might cause SolidWorks to fail."
[image]

nallath, 2017-09-13:
The issue has been fixed for the next release.

churusaa, 2018-06-22:
This seems to still be an issue in 3.2.1 on Linux. Open Cura with webcam stream active to monitor will eventually consume all free pages in memory and cause the system to become unresponsive.

cura 3.2.1 (PPA)

4.15.0-23-generic #25-Ubuntu SMP Wed May 23 18:02:16 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

nvidia G98 [GeForce 9300 GE] graphics with Nouveau drivers: xserver-xorg-video-nouveau 1.0.15-2 amd64 and libdrm-nouveau2 2.4.91-2 amd64

lscpu:
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  2
Core(s) per socket:  2
Socket(s):           1
NUMA node(s):        1
Vendor ID:           GenuineIntel
CPU family:          6
Model:               60
Model name:          Intel(R) Core(TM) i3-4130 CPU @ 3.40GHz
Stepping:            3
CPU MHz:             3392.394
CPU max MHz:         3400.0000
CPU min MHz:         800.0000
BogoMIPS:            6784.66
Virtualization:      VT-x
L1d cache:           32K
L1i cache:           32K
L2 cache:            256K
L3 cache:            3072K
NUMA node0 CPU(s):   0-3
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm cpuid_fault epb invpcid_single pti ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm arat pln pts

Please let me know if more (or less) information is required and what I can do to help track this down.

Ghostkeeper, 2018-08-07:
We haven't been able to reproduce that so far. At least not on 3.4.