Ultimaker/Cura #2381: Postprocessing pluginw with cura 2.7

JPFrancoia, 2017-09-04:
Hi,

Are the PostrProcessingPlugins operational with cura 2.7 ? It seems not at the moment.
Cura installed on Linux.

I cloned the PostProcessingPlugin repo in ~/.local/share/cura/master/plugins, but the menu doesn't appear.

Cheers

JPFrancoia, 2017-09-04:
Ah, I found the issue.

Before cura 2.7:

```
.local/share/cura/master/plugins/PostProcessingPlugin
```

After cura 2.7 it needs to be:
```
.local/share/cura/master/plugins/PostProcessingPlugin/PostProcessingPlugin
```

And it also needs a ```plugin.json``` file, it doesn't have one now.

```
{
	"name": "PostProcessing plugin",
	"author": "John do",
	"version": "2.2",
	"api": 4,
	"description": "Post processing"
}
```

It's probably not the right json, and also the author of the plugin doesn't get credit, but sorry I was in a rush.

Also, maybe you could improve the "what's new" (like putting a link ?) when compatibility is broken like that ?

fieldOfView, 2017-09-04:
You probably got the wrong branch of the PostProcessingPlugin. You need the 2.7 branch or the master branch. Those come with a plugin.json file.

JPFrancoia, 2017-09-05:
True.

closed 2017-09-05 by JPFrancoia