Ultimaker/Cura #2383: Possible issues with multiple cersions of Cura installed

MFHFozzy, 2017-09-04:
Ive had issues with 2.6, 2.7Beat and 2.7 release.
Ive filed bugs on all 3.
When i noticed that 2.7 was released, i uninstalled the other 2 and installed 2.7
I was going to test the old bugs on 2.7. But it would not launch.
The process would start...and they just quit a few seconds later (~10-15).
No dialog, nada. 
So i re-installed 2.7 Beta. Now *it* doesnt run either. But it *does* show a crash report. (listed below)
So i went back to 2.6 and *it* finally runs.
All 3 are installed to different directories.

I looked at the logs for 2.7 release after trying to launch it and saw this:
2017-07-27 17:08:25,869 - INFO - UM.Qt.QtApplication.__init__ [112]: Command line arguments: {'qmljsdebugger': None, 'disable-textures': False, 'single_instance': False, 'file': [], 'external-backend': False}
2017-07-27 17:08:25,878 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-07-27 17:08:25,881 - DEBUG - UM.Backend.Backend._logSocketState [168]: Socket state changed to Listening
2017-07-27 17:08:25,902 - INFO - UM.Backend.Backend.startEngine [72]: Started engine process: D:\Apps\Cura 2.6\CuraEngine.exe
2017-07-27 17:08:25,904 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] Calling engine with: ['D:\\Apps\\Cura 2.6\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'D:\\Apps\\Cura 2.6\\resources\\definitions\\fdmprinter.def.json', '']

Why is 2.7 is calling the engine from 2.6??!?!?! 
[cura - Copy.zip](https://github.com/Ultimaker/Cura/files/1275480/cura.-.Copy.zip)

2.7 Beta crash report:
Version: 2.7.0-BETA
Platform: Windows-10-10.0.15063
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\2.7\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24, in <module>
  File "D:/2.7/build/inst/bin/cura_app.py", line 73, in <module>
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 646, in run
AttributeError: 'NoneType' object has no attribute 'setOrigin'

-------------------------
Thread 0x0000526c (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 155 in _storeStderrToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x0000299c (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 147 in _storeOutputToLogThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00001f10 (most recent call first):
  File "D:\Apps\Cura 2.7 Beta\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "D:\env\2.7\inst\lib\threading.py", line 862 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00004900 (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 293 in wait
  File "D:\env\2.7\inst\lib\threading.py", line 424 in acquire
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.7/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x0000120c (most recent call first):
[... 89 lines cut ...]
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000040d8 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x000030dc (most recent call first):
  File "D:\env\2.7\inst\lib\threading.py", line 297 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Thread 0x00004b28 (most recent call first):
  File "D:\env\2.7\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.7\inst\lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.7\inst\lib\threading.py", line 882 in _bootstrap

Current thread 0x00000d98 (most recent call first):
  File "D:/2.7/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook

MFHFozzy, 2017-09-04:
Running the uninstall for 2.6 and 2.7 did not remove all folders and files.
Local and roaming still had cura folders, and the program directories still had the plugin folders. 
Uninstalled *everything* again. 

Deleted all cura fodlers i could find in %homepath%\appdata\local\cura and %homepath%\appdata\roaming\cura
emptied all temp folders.
then re-ran install of 2.7 release.

Vandrasc, 2017-09-05:
I am able to reproduce the same error when I manually disable the Camera Tool in the Preferences->Plugins window. 

I could successfully launch Cura again after removing my user directory from C:\Users\<username>\AppData\Roaming\Cura\Cura 2.7. Are you sure all the files have been removed?

MFHFozzy, 2017-09-05:
Oops sorry, meant to say that after deleting *everything* Cura started working again.

Ghostkeeper, 2017-09-11:
It could use the old CuraEngine if the preferences file was changed to specifically point to the engine location. You'd have to change that manually with a text editor though.

MFHFozzy, 2017-11-04:
Point was/is that the uninstall of the older versions left files around that screwed up the newer versions.
Anyone looking at the uninstaller?

MFHFozzy, 2017-11-04:
Err testing, i should say.

closed 2018-03-02 by ChrisTerBeke

Ghostkeeper, 2018-03-26:
The issue with the uninstaller is reported here: https://github.com/Ultimaker/Cura/issues/2359