Ultimaker/Cura #2384: Cura 2.7 loses network settings when multiple printers are present

echo465, 2017-09-05:
I have 3 printers at 192.168.0.101, .102, and .103 all running Octopi from a Raspberry Pi, all on port 80.  Occasionally, when I submit a job from Cura to Octoprint, I get the message 'Octoprint is offline. Unable to start a new job'.  When I go to manage printers, I see that the printer isn't connected and the Octoprint instance now shows the IP address of one of the other printers, but with the current printer's  API key, with the message 'The API key is not valid.'

I've had this happen 2 or 3 times since I updated to 2.7.

fieldOfView, 2017-09-05:
Could you try adding 3 manual octoprint instances instead of using the automatically discovered ones? If the automatically discovered ones disappear off the list, that indicates an issue with Bonjour/ZeroConf on your network.

Logs couldbe useful. Could you paste a recent section to pastebin.com?

ChrisTerBeke, 2017-11-14:
Is this still an issue?

closed 2018-03-09 by ianpaschal