Ultimaker/Cura #2385: Master branch lagging after released version

maukcc, 2017-09-05:
The released version at this time is 2.7, master branch is at 2.6.99
How is it possible that the master branch (which should be used for alpha testing) is lagging the released version?

nallath, 2017-09-05:
master just uses "master" as a version number. At least, that what it's doing for me.

smartavionics, 2017-09-05:
I think that must be the PPA, the OP is referring to.

nallath, 2017-09-05:
Ah, well the PPA is not officially supported as @thopiekar does that in his own free time. So it can occasionally be a bit behind.

thopiekar, 2017-09-05:
Yep, and I'm too busy to do the update currently.

maukcc, 2017-09-05:
Ahh, I see now. Your master branch is a couple of commits behind :) the ultimaker one.
No worries, thanks for your efforts anyhow thopiekar, I am loving it.

maukcc, 2017-09-05:
this can be closed

closed 2017-09-05 by nallath