Ultimaker/Cura #2386: Needs reslice after saving profile

smartavionics, 2017-09-05:
Not a big deal this but I find it slightly irksome that if you make some changes to a profile, slice, and then save the profile, it thinks the profile has been changed and requires you to re-slice before you can save the gcode.

Ghostkeeper, 2018-01-01:
Our project manager just removed this issue from our planning because it's older than 12 weeks.

smartavionics, 2018-01-01:
Yeah, well, the bug still exists and (AFAIK) no one has made any attempt to fix it. If I knew how the UI functioned I would have a go myself but I don't so I am hoping that someone else will.

Ghostkeeper, 2018-01-01:
Yeah the reason is that we reslice when a profile is switched. When you save your current changes the program creates a new profile, puts all your current changes in there and activates the new profile. The last two steps cause the reslice. The steps of moving your current changes to a new profile (clearing your current changes) and activating the new profile both individually require a reslice, but it doesn't realise that as a whole the settings should remain the same.

smartavionics, 2018-01-01:
Understood. In the grand scheme of things, it's a minor annoyance. There's far more important stuff that needs fixing first, I'll just ignore it.

closed 2018-01-01 by smartavionics