Ultimaker/Cura #2387: Not 100% since cura 2.7

JPFrancoia, 2017-09-05:
Hi,

Since the update to 2.7, I don't get the behaviour I had with previous versions of cura.

I set the infill to 100% and the infill overlap percentage to 100% (I print PP and I want my prints to be leak proof). Before, the infills were printed at 100%, like the walls, no air, no infill pattern.

Since Cura 2.7, I can see that the infills are printed with a cubic pattern. There is air between the lines of the infill, it's not printed at 100%. I didn't change my profile.

What changed ? I can see a new setting now: gradual infill steps (set to 0 for now). Is it responsible ?

Cheers

JPFrancoia, 2017-09-05:
Ah, again, I found the issue. Even if the infills are set to 100%, if infill pattern is set to cubic or grid, the infills won't actually be at 100%.

mikeq3, 2017-09-08:
You can get water tight by changing your top and bottom line width to less then 0.4 nozzle size or whatever you nozzle size is. 20 - 25% does a good job. 0.4 - 20% = 0.32
This makes the top and bottom fill lines tighter together.

JPFrancoia, 2017-09-08:
ah, good to know. What about the walls ? I don't think I can change the width of the line for the shell.

mikeq3, 2017-09-09:
Check the invisible settings and enable wall and top and bottom line width. I would leave infill alone. With those reduced by 20 - 25% you should get tight line overlap. You can use a usual 30 % infill with normal line width.

Ghostkeeper, 2017-09-11:
The infill is in theory still 100% even with grid or cubic patterns or whatever. There's a lot of overlap between lines and if you assume that this overlap oozes over into the adjacent holes it would have 100% infill. In practice, if the infill lines overlap, it will not be able to extrude as much as you'd like so the infill will not be 100%.

If you want 100% infill, I advise to just use lines or fill everything with skin (as is the default behaviour).

JPFrancoia, 2017-09-12:
@Ghostkeeper Thanks, now I understand.

@mikeq3 : I have several questions/comments (maybe it's better if we open a new issue ?)

- What physically happens when you reduce the wall line width ? For example, I have an 0.6 nozzle, I reduced the width to 0.48 mm. What will happen in real life ? Will the printer print 0.48 lines, or will it think it's printing 0.48 lines, but really printing 0.6 mm lines (resulting in more lines printed) ?

- Do you have any experience printing PP ?

Here is what happened when I changed the wall/top/bottom line width (20% reduction):

[image]

Settings:

[image]

It's not water tight. I can get it water tight with these settings:

[image]

The difference is that infill percentage, infill overlap percentage and skin overlap percentage are set to 100%. Material flow is also set to 150% (over extrusion).

Do you have any comment/suggestion ? That much over extrusion, and such a high percentage of infill make my parts hard to print.

DDDirk, 2017-09-13:
What do you mean by 'hard to print'? Are the dimensions wrong? I don't know what Cura does with those settings, but I once printed some tensile bars (programmed, not sliced) with an extrusion factor of 200% (so 100% overextrusion) and the bars were almost completely dense (I weighed them). The dimensions increased slightly, though, but that might be solved by using overextrusion for the inner walls (as many as you can fit in actually), but not for the outer wall.
Also, it might be advisable to print slower, because with an increased extrusion rate (feed rate) the extruder might find it harder to keep up. So the extrusion rate should preferably be kept the same. So with an extrusion factor of 150%, set the print speed at 2/3 (100/150).

Actually, just reducing the print speed might already make a difference. This may also improve the print in other areas. But that is not for the impatient.

JPFrancoia, 2017-09-13:
By hard to print I mean that cracks are more likely to happen (the part is basically one big piece of dense plastic): the layers often detach from each other.

Also if you look closely at the settings, I'm already printing at 15 mm/s (that's slow). I tried many different speeds: at 20 mm/s, the part starts leaking.

DDDirk, 2017-09-13:
I have no experience with PP, but I understand it's not the easiest of materials.
You print at a layer height of 0.3 mm, which is rather much. You'd expect that to be easier with a wider nozzle, but that actually worsens things because both influence the extrusion speed, which is:
layer height * road width * print speed * flow rate
In your case
0.3 * 0.6 * 15 * 1.5 = 4.05 mm³/s
Which is rather much, meaning you likely don't actually get a flow rate of 150%. Normally that is not a big problem, but if you want the print to be watertight it may very well be.
So the easiest solution would be to reduce the layer height.
Lowering the print speed even further might cause problems with the overextrusion (flow rate). With those tensile bars I mentioned, I also once combined 200% flow with a print speed of 5 mm/s, which caused actual overextrusion, with the material piling up at the sides. I think the reason was that the previous layer melted, giving the overextrusion more room.  Normally the material can only go sideways a bit, giving wider roads (which was desirable in my case as it is in yours). In other words, you don't get actual overextrusion (except at the outer walls), just maximum extrusion and density.

mikeq3, 2017-09-13:
JPFrancoia
Yes printing 0.48 lines with a 0.6 nozzle will cause the 0.6 lines to tightly overlap. You wil have more lines as well. The infill overlap setting is for infill line end overlap.  Not side to side.  Side to side line overlap can be achieved reducing line width. You shouldn't need to increase extrusion.

Ghostkeeper, 2017-09-15:
> What physically happens when you reduce the wall line width ?

Cura will draw paths with enough extrusion for a smaller (0.48mm) line, and draw those lines closer together. Since the actual nozzle opening is still 0.6mm, the line will be slightly less regular. We've also found that setting the line with to about 7/8ths of the nozzle size creates better bonding between two adjacent lines for this reason. Basically you're extruding partially on top of the adjacent line, so the bonding is very good.

> (About infill percentage, infill overlap percentage and skin overlap percentage) I don't know what Cura does with those settings

Infill percentage will cause infill lines to be printed closer together. Most infill patterns are based on adjacent lines. In the simple case (Lines infill) the lines will be touching exactly at 100% infill, but you can go above 100% to produce lines that overlap, so you get overextrusion. For some patterns, such as grid, the line distance is multiplied by 2 since there are 2 directions for the lines. For triangles it's multiplied by 3, etc. Infill overlap percentage is a percentage of the inner wall line width that the infill should overlap with the innermost wall. It will overextrude a bit. It is meant to make the infill adhere to the inner wall better. Skin overlap percentage is a similar setting for skin lines to better adhere to walls.

In my experience, you should treat PP similar to TPU: It's quite flexible, so if you have a Bowden tube like the Ultimaker printers you should prevent retractions and other interruptions in flow as much as possible.

JPFrancoia, 2017-09-18:
Thank you all for your answers. So to summarize:

- Reduce line width for the walls (7/8th or 80% ?)
- Increase infill percentage to 100%+
- No need to set the material flow to 150%

@Ghostkeeper : I just installed a E3D titan extruder, in a direct extrusion configuration. Do you think it will perform better than a normal bowden extruder ?

@Ghostkeeper Should I disable retraction for PP ?

JPFrancoia, 2017-09-19:
Ok I'm closing the issue since I got the answer to my first question. Feel free to continue answering though.

closed 2017-09-19 by JPFrancoia

Ghostkeeper, 2017-09-20:
> I just installed a E3D titan extruder, in a direct extrusion configuration. Do you think it will perform better than a normal bowden extruder ?

Definitely, for flexible materials at least. The major advantage of Bowden tubes is that you don't place the motor/extruder on the print head, which greatly reduces weight and therefore increases the print speed you can get. You'd lose that, but the advantage of being able to retract properly is great.

> Should I disable retraction for PP ?

Probably not. But the retraction distance needs to be fairly big. I'd also try to choose settings that reduce the number of retractions required, such as zig-zag infill (or Cross infill in the upcoming release).