Ultimaker/Cura #2389: change filament/extruder, determine percentage/layer with two or more extruders

maaspt, 2017-09-06:
Hi

I upgraded to two extruders one nozzle, any option to change extruder in determine layer/percentage ?

In Post processing I can Pause at Heigth I choose, and change filament manually in the same extruder, but with two or more extruders is there any function to change them automatic ?

Just for changing layers colors in some projects

Regards

nallath, 2017-09-06:
We don't support anything like that at the moment.

maaspt, 2017-09-08:
Thanks for your answer

It would be o good feature for more creative and colorful prints

Ghostkeeper, 2017-09-15:
Is there some g-code we can output to implement that? I don't know of any "change material" command in g-code.

Looking it up, I discovered M701 and M702 as candidates: http://reprap.org/wiki/G-code#M701:_Load_filament
But that's only supported by RepRap apparently.

maaspt, 2017-09-18:
I can not find any command for that.

Ghostkeeper, 2017-09-24:
Me neither :( I think this is then limited by the hardware and firmware support.

ChrisTerBeke, 2017-11-14:
You can put @pause commands in G-code, but that will only work if the host driver software has support for that. Afaik there are no firmwares that do this by themselves.

Ghostkeeper, 2017-11-22:
We now have a post-processing plug-in that pauses and commands the printer to change material:
https://github.com/nallath/PostProcessingPlugin/pull/49
It'll be released with Cura 3.1. This makes use of the M600 command.

This is not exactly what the user requested though. He just wanted a tool change on his dual-extrusion printer. In that case I think he should just add a cube to his build plate, enable the "Cutting Mesh" per-object setting on that cube, and then make it intersect with his normal model at the height that he desires. That achieves what Maaspt wanted, though it's not very user-friendly.

closed 2017-11-22 by Ghostkeeper

fieldOfView, 2017-11-22:
> In that case I think he should just add a cube to his build plate, enable the "Cutting Mesh" per-object setting on that cube, and then make it intersect with his normal model at the height that he desires. That achieves what @maaspt wanted, though it's not very user-friendly.

See #2758 though.

maaspt, 2017-11-23:
I`am going to try it

Thanks