Ultimaker/Cura #2390: Concentric ironing order

AmbientChaos, 2017-09-06:
I have been doing some experimentation with the new ironing feature, and with the concentric ironing pattern I have noticed one thing that we can't change but has a significant effect on the surface finish.  Sometimes the concentric rings spiral _out from_ a central point, and sometimes they spiral _in toward_ a central point.  When a set of concentric rings converges down to a small point or line, starting from the middle and working outward gives a clean surface finish,
[image]

 but starting from the outside and working inward results in a raised blob forming in the middle of the concentric rings.
[image]
The process and pathing for the ironing seem to work great, it's just the order of operations that is sometimes giving undesirable results.  Would it be possible to make this behavior less random, like if the central concentric ring is below a certain length then it will always start at the inside and work outward or something?

LipuFei, 2017-09-07:
Hi @AmbientChaos , according to my experience, the zig-zag pattern for the ironing seems to produce better surfaces in many cases. Have you tried that too?

@Ghostkeeper has more insight on the ironing feature. He will reply to you when he's back from vacation.

Ghostkeeper, 2017-09-13:
This is actually quite an interesting result.

I've seen the blob in the middle with concentric as well. I just accepted it as being inherent with concentric, but if as you say the concentric from inside to outside does work, then we could consider implementing that as the default behaviour.
In fact, we could consider making that the default behaviour for normal skin as well.

Normally it would almost always go from outside to inside, because it starts with the closest line which is usually the outermost one.

AmbientChaos, 2017-09-13:
The two pictures  are the mirrored sides of the same part, so the print settings are all the same.  I just happened to be curious enough about the ironing to watch it do one side after the other, working outward for the first and inward for the second.  Now that you mention it, since I was also recently trying the concentric pattern for the top layer of normal skin as well, I believe that I noticed much the same phenomena for normal skin, it's just somewhat less noticeable than it is against the ironed surface.

Ghostkeeper, 2017-12-11:
This just got removed from our planning by our project manager because it's older than 12 weeks.

I've made a test branch to try it out: `feature_infill_inside_out`. I'm making a few test prints and if the result is better I plan to just merge it or make a PR. It's very easy to implement.

dstulken, 2019-07-19:
Noticed this old bug report was still open...
I would agree that inside-out concentric ironing would be a "safe" default behavior, and very much appreciated for anyone ironing with concentric patterns.
  
 @Ghostkeeper, was your test branch code ever merged into the mainline?

Ghostkeeper, 2019-07-22:
Yes, it was merged in Cura 3.3. This issue is fixed now, so I'll close it. It's always printing concentric patterns from the inside out.

Commit that fixed it: https://github.com/Ultimaker/CuraEngine/commit/c86501427571f2d2879be7aa994e9b38615bd391

closed 2019-07-22 by Ghostkeeper

print-it, 2021-02-16:
Hi,
I really need your help.
I am useing Cura 4.8 (downloaded from the Ultimaker website) and have the same issue when ironing the top-layer in a concentric pattern just like in the pictures above.

@Ghostkeeper 
Do I have to edit the programmcode by myself or are the changes that you mentioned already implemented in the default 4.8 Version ? (sorry for asking, I am very inexpierienced with programming)

Thanks!

Ghostkeeper, 2021-02-24:
Those changes were made to 3.3, so the issue should still be fixed in 4.8. When I load in a simple model, enable ironing and set the top/bottom pattern and ironing pattern both to Concentric, it seems to correctly print the pattern from the inside out, both for skin and for ironing:

[image]

Do you have a project file that reproduces the problem? Perhaps it'd be good to open a new issue about it.