Ultimaker/Cura #2392: Retraction extra prime amount negative value not possible

konvoj, 2017-09-06:
For some printers or print modes a way to eliminate oozing or overextruding of material is to use parameter Retraction Extra Prime Amount having negative value, for direct drive printer and PETG, I'm using -0.04 mm of length. This is how Slic3r works, similar effect can be achieved in S3D.

Cura is handling things differently, first of all the value is in mm3 (not a big deal), second by using the negative value the actual value stays at 0, in the final gcode you can see clearly that the retraction and insertion distance is the same.

Eganwp, 2017-09-11:
+1. A negative value is having no effect for me either, leaving bumps and pimples on the start of all extrusions on layers. Really need this fixed in order to clean-up prints!

nallath, 2017-09-11:
For our reference; CURA-4315

nallath, 2017-09-11:
Fixed it in 27636106f4f298332f9220513b69a975370eda53 (no idea why we introduced this in the first place. It makes no sense to force it to be positive)

closed 2017-09-11 by nallath

HClO4, 2018-02-10:
Sadly in the new Cura 3.2 the value still has to be positive...

devils27, 2018-02-26:
STILL NOT FIXED.  seems like a pretty simple fix in the code.  why has this been 3 releases so far and still not fixed?