Ultimaker/Cura #2394: Scale sinks model into printbed

SuperDoxin, 2017-09-06:
Application Version: 2.7
Platform: Debian Linux
PyQt: not relevant.
Display Driver: not relevant.

Steps to Reproduce:
Import a tiny model, Scale it up significantly (e.g. 5000%)

Actual Results:
The model sinks into the printbed a fair few mm.

Expected results:
The model doesn't sink into the printbed more than a layerheight or so.

Additional Information:
Screenshot: http://i.imgur.com/0IepjDJ.png
Using the move tool doesn't help. The model snaps back into its original place.

Vandrasc, 2017-09-07:
Hi, is the Automatically drop models to the build plate option enabled? You can verify that in Preferences -> Configure Cura -> General and under the Viewport behavior. If it is enabled, this issue shouldn't occur. Thanks!

SuperDoxin, 2017-09-07:
If that option is enabled by default it is still enabled. Do mind that the models all do drop down to the build plate, they just end up slightly too low, getting worse as the scale gets higher.

Like I said, if I drag the model up using the move tool it'll just snap back down to it's sunken position. Scaling the STL file outside of cura makes the problem go away so there's definitely a bug in cura involved.

smartavionics, 2017-09-07:
So what happens if you disable the drop onto build plate option and then manually set the z position to 0?

Thisismydigitalself, 2017-09-07:
+1
Same here.  scale up small object, then scale it back down and it wont snap to build plate but rather hang in mid air high above bed. once clicked on it will drop down to bed.

SuperDoxin, 2017-09-07:
@Thisismydigitalself That's a different issue though. My problem is the model being sunk into the bed when it is scaled up.

@smartavionics Presumably I'd be able to drag the model up, but that's still not fixing the bug with the drop onto build plate mechanism.

smartavionics, 2017-09-07:
> @smartavionics Presumably I'd be able to drag the model up, but that's still not fixing the bug with the drop onto build plate mechanism.

True, but i'm just trying to understand if it really does know where the bottom of the model is. When the z offset is set to 0, if the model is just touching the build plate then at least Cura has that much right!

Ghostkeeper, 2017-09-15:
At the moment, the dropping on build plate seems to be broken altogether (in Master)!

RudiKlein, 2017-10-15:
Hi,
I just ran into this thread for another issue and did some testing. It appears that the failure to land on the bed properly occurs when the size of the model is larger then the bed. As soon as you shrink the enlarged model to fit in the bed area, it drops down on Z 0.0

For what it's worth.

Bye,

Rudi

SuperDoxin, 2017-10-15:
The issue happens at large scale factors, even if the model is smaller than the size of the bed.

Ghostkeeper, 2017-10-22:
If it's larger than the build volume, not dropping down is expected behaviour. If you do align the bottom of the model to the build plate, then in many cases you're not even able to see the model because it's been moved so far up, if it doesn't happen to have any part around that tiny cube/cylinder that's now your build volume.

This issue was about models that are so tiny that their scale factor needs to be huge but still fit in the volume.

I created a model file that's tiny and needs to be scaled to 10.000% for normal size. But it doesn't reproduce the problem. See attachment: [PrikbordHaakjeTiny.zip](https://github.com/Ultimaker/Cura/files/1405066/PrikbordHaakjeTiny.zip)
It seems to align perfectly fine to the bottom.

closed 2018-03-02 by ChrisTerBeke