Ultimaker/Cura #2395: Head hit the print at layer 32 - Cura 2.7 on Win7

marrincia, 2017-09-06:
Hello,

the generation of the Gcoda was wrong at 6.7mm height (Layer 27).

The printer head crashed into the part, moving the glass plate against the wall and causing the fan shroud to bent.

Before understanding the problem i thought it was an hardware issue but checking the code, after running anther print that stopped in the same way (luckly I was at home this time) I found this mistake at layer 32. the height jump to lower level at the next layer and it seems the code is overwritten by another as there is the start of a file generated with cura 2.5.0

UM2-C_Logo_.gcode

I hope the harware of my printer is not damaged....

smartavionics, 2017-09-06:
Could you make both the model and the gcode available please?

marrincia, 2017-09-07:
Hello,

I tried to drag and drop the .gcode in the window but it doesn´t allow me to do it.
Any advice? (sorry it´s the first time i have to report something on this platform).

I will also try to include the .stl file as soon as I´m back home.

Thinking about what happened, it seems like there is a code of another print in the middle of this file...it is possible that overwriting an existing file something went wrong?

I´m currently at work, but as soon as I´m back home i will try to forward you all the data!

It was a pretty scary experience and i would like this to happen again :(...
and hopefully the printer is not damaged...

thanks in advance for your help

smartavionics, 2017-09-07:
Hi, I don't use the drag and drop but you could also try the "selecting them" link that's at the bottom of this edit box. Hmm, I just tried that and it wouldn't let me upload an STL file but it does allow ZIP files so just zip the STL and upload that.

marrincia, 2017-09-07:
[Archive.zip](https://github.com/Ultimaker/Cura/files/1284469/Archive.zip)

here you go :)

.stl and .gcode

thanks a lot

smartavionics, 2017-09-07:
I think you're right, somehow, the gcode for two prints have become merged. The original STL file looks fine. Maybe there's a problem with the disk or the system crashed and the two files got somehow corrupted?

[image]

marrincia, 2017-09-07:
wow!

yes this is an old project I worked on weeks ago!!

I will definetely checke the card and/or scrap it.
Really dangerous behaviour fot the printer.

Could I ask how you managed to open the gcode in this format? 
I would like to use this check for future important prints.

Many many thanks fr your help!!

smartavionics, 2017-09-07:
I tried to open it with Cura but it failed so I used Simplify3D which I no longer use for slicing but it still has a better gcode viewer than Cura does. I tried another program and that couldn't load it either so I think that S3D can just cope with broken gcode best.

zerspaner-gerd, 2017-09-08:
From line 143188 many errors occur.

"CraftWare" also has a good gcode viewer, and is free.

[image]

marrincia, 2017-09-08:
yes it was pa big mess...

thanks a lot for your feedback

smartavionics, 2017-09-09:
I just tried out the Linux version of craftware and it has a strange problem in that it complains about bad gcode syntax when loading a Cura generated gcode file. It always appears around the same point in the file and there's actually nothing wrong at that point in the file. Nice gcode display, though.

smartavionics, 2017-09-09:
And it slices really quick!

Ghostkeeper, 2017-09-15:
Can we assume this was due to memory or disk corruption? I don't see how we would be able to reproduce this problem.

ianpaschal, 2018-03-09:
I'm closing due to age and irreproducibility.

closed 2018-03-09 by ianpaschal