Ultimaker/Cura #2396: Cura 2.7.0 - Number of Layers UI issue

amagr0, 2017-09-07:
Incorrect space for the number of layers in "View Mode: Layers" when are less than 100 layers.

[image]

[image]

ChrisTerBeke, 2017-09-27:
As this interface element is completely changed in the next release, closing this for now. Please re-open if the upcoming release still gives you issues with this.

Ghostkeeper, 2017-10-02:
This still occurs:
[image]
I loaded an 1cm cube with UM3's default settings.

PR #2540 (2017-10-02): Fix width of the layerview layer box
This PR changes the width of the layerview layer number indicator.

Fixes #2396
commit 50f596cc (2017-10-02): Fix width of the layerview layer box

Fixes #2396
closed 2017-10-02 by Ghostkeeper