Ultimaker/Cura #2397: Cura 2.7.0 - Multiply group error

eloipr, 2017-09-07:
I'm running the Cura 2.7.0 AppImage on Ubuntu 16.04.
When I try to multiply a group I get the next error:
```
2017-09-07 15:40:04,279 - ERROR - UM.Logger.logException [76]: Exception: Job <cura.MultiplyObjectsJob.MultiplyObjectsJob object at 0x1891c8d8> caused an exception
2017-09-07 15:40:04,282 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-09-07 15:40:04,285 - ERROR - UM.Logger.logException [80]:   File "/build/2.7/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 125, in run
2017-09-07 15:40:04,288 - ERROR - UM.Logger.logException [80]:   File "/build/2.7/build/inst/lib/python3/dist-packages/cura/MultiplyObjectsJob.py", line 53, in run
2017-09-07 15:40:04,291 - ERROR - UM.Logger.logException [80]:   File "/build/2.7/build/inst/lib/python3/dist-packages/cura/ShapeArray.py", line 49, in fromNode
2017-09-07 15:40:04,294 - ERROR - UM.Logger.logException [80]: AttributeError: 'NoneType' object has no attribute 'getMinkowskiHull'
```
It looks like the group SceneNode doesn't have a ConvexHullDecorator.

nallath, 2017-09-07:
Odd, I just tried this and it works. Could you post some reproduction steps? 
I did the following;
1. Add 1 object
2. duplicate it
3. group both objects
4. Multiply the group

eloipr, 2017-09-08:
I've done the same steps as you.
I get the error when after grouping both objects I multiply the group without unselecting it. If after grouping the objects I unselect the group (click somewhere else) and then multiply it, it works.

I've done some debugging and when I unselect the group, the method __onChangeTimerFinished_ in PlatformPhysics.py is called and a ConvexHullDecorator is added to the group node. Then multiply works.

nallath, 2017-09-11:
Ah, it could be that i did unselect them in between. But if you've already found a fix, could you make a pull request for that? We also use the pull requests to give credits in our changelogs.

eloipr, 2017-09-12:
Ok, pull request created!

LipuFei, 2017-09-12:
Closed via #2418

closed 2017-09-12 by LipuFei