Ultimaker/Cura #2400: 2.7 Beta worked new 2.7 does not

PapaG-Design, 2017-09-07:
Uploaded latest version replacing the Beta version and now the program will not open goes through the opening process get to preparing scenes and then nothing loading screen goes away.  Tried reloading it same issue.  2.7 Beta was working great.

Thisismydigitalself, 2017-09-07:
Try renaming cura 2.7 folder to 2.7BETA and reinstall 2.7 which will create a fresh folder.

PapaG-Design, 2017-09-07:
Did not work.

<https://github.com/Ultimaker/Cura/issues/2400#issuecomment-327856721>, or mute
the thread

MFHFozzy, 2017-09-08:
try this:
Uninstall everything.
Then Delet all cura fodlers in
%homepath%\appdata\local\cura 
and
%homepath%\appdata\roaming\cura
Then Empty all temp folders.
Then re-install 2.7 release.

They dont do a good job when uninstalling.

Ghostkeeper, 2017-09-15:
Renaming the file didn't work properly because it then found a 2.6 folder and upgraded those again.

To actually fix this, I'd like to see a log file of when you open Cura 2.7.

ianpaschal, 2018-02-27:
No info was provided after several weeks, so I'm closing this.

closed 2018-02-27 by ianpaschal