Ultimaker/Cura #2401: "Z-Hop only over printed parts" does not hop over supports

dpy-, 2017-09-08:
Cura 2.7 (or earlier, not tested with earlier versions) does not generate a z-hop over supports when this option is selected.  This may be working as intended, in which case another checkbox to also hop over supports would be nice.  I'm having issues printing a model with supports where the head is knocking the supports over if I have this option selected.  Using Prusa i3 Mk2s.

Thisismydigitalself, 2017-09-08:
Following your post I started using Z hop which i never used before.
Me like! very clean prints. I will check your claim though..

Thisismydigitalself, 2017-09-08:
2.7 hops over supports as well.
I kinda like this feature.. i did not use it till now and it works great.

dpy-, 2017-09-08:
I made a small test print to double check the behavior.  In this print, there are two walls and a roof, so there are a few travels over the support for the roof (see attached screenshot).  I generated gcode with "zhop only over printed parts" off (ZHopTest1.gcode) and on (ZHopTest2.gcode).  You can see in ZHopTest2 that there are no Z movements while printing the Wall-Outer section of each layer, but in ZHopTest1 there are.
[image]
[ZHopTest1.gcode.txt](https://github.com/Ultimaker/Cura/files/1289301/ZHopTest1.gcode.txt)
[ZHopTest2.gcode.txt](https://github.com/Ultimaker/Cura/files/1289302/ZHopTest2.gcode.txt)

Thisismydigitalself, 2017-09-09:
My Settings:
[image]

Ghostkeeper, 2017-09-15:
As far as I know, Z Hop should _not_ hop over support by default. We generally assume that support doesn't need to be pretty. We will hop over support if the hop also hops over the printed model anyway (when combing is disabled), but not when moving from support to support.

Ghostkeeper, 2017-09-15:
Why would you want to Z Hop over support?

KeithAZ3d, 2018-02-18:
Printing with plastics that have a tendency to warp will do so more prominently in the support areas because by default the plastic is laid down faster.  So, in turn, your printhead will ultimately collide with the supports causing huge drastic problems later on in larger prints.  This has been my number one issue to not have just Z-hop over every little printed thing within a certain printed time frame.  So I am forced to either print supports to match speeds as my outer perimeter and have a 20hour print stretch out to 4 or 5 days or I throw a hail Mary and do what everybody thinks is best, which is always confusing. Please correct me or please share any advice that I may have missed something or can try different support tweaks to divert this problem

Ghostkeeper, 2018-03-09:
The warping by itself is an argument. Let's see what people think of this.

Ghostkeeper, 2018-05-07:
Our project manager put this off for later. We'll not get the time to implement this any time soon.

picaschaf, 2018-05-14:
I had to find this issue to find out why Z-Hop doesn't work. For us it's not about prettyness (no problem with that at all), but the Nozzle constantly crashing against infill or support and ruining it is a nogo. For me this issue would be a P1 issue.

VanessaE, 2018-05-17:
In 3.3.1, it appears to not hop over the part at all most of the time - a horizontal move is performed whenever possible, even if the result would wind and meander around creating a much greater travel time than a straight line with hop (on some bots, Z moves are fast enough to not really matter).

Maybe what's needed is for Cura to prefer to always hop to avoid the part (if enabled, that is), if both ends of the travel are somewhere that will hide the artifacts that form due to hopping (e.g. infill, inner skin layers, inner perimeters that aren't on a top layer).  It'll always be shorter than trying to just avoid the part, but horizontal-only moves should be performed if either end of the move can't be hidden.

JohnEdwa, 2018-08-05:
I'd like to add my support for this as well, even if the reason so far is only that the few prints I've had to do multiple times have had a really long straight support section that the print head will travel right through, and without a Z-hop or avoidance it sounds like [the hotend trying to play a washboard solo every 25 seconds. ](https://www.youtube.com/watch?v=TAexBL5eToE)

FeepingCreature, 2019-01-07:
I have the exact same issue. I would love to have some way to avoid supports when travelling so I can not have the printer rip apart the supporting structure that I need to properly print overhang. Support isn't _optional_, guys.

LuckierDodge, 2019-02-17:
Can we get any sort of eta on this? This doesn't seem like something that would be too much trouble to implement, and it would make some big challenges I'm having right now pretty much trivial

Ghostkeeper, 2019-02-18:
> I would love to have some way to avoid supports when travelling

You could always try the "Avoid Supports When Traveling" setting. If you can't avoid support (because you're surrounded) it will not z-hop though.

> Can we get any sort of eta on this?

No, this was decided to be too much trouble to implement for what it's worth. I guess this was because you don't typically knock over support when you are surrounded by it (because it's typically quite big then).

picaschaf, 2019-02-18:
Knocking over support is really no issue for us, but toplayer look like crap when the head homes back. All other printer that I know just lift Z for that reason.

adamsiem, 2020-04-02:
@Ghostkeeper I need 'Combing Mode: All' to ensure my prints comb for supports. Currently, they do not. As you can see in my attached screenshot - my large 7.75 hour print combs for all rows BUT a few support lines in a slot. This means if there is slight warping, the extruder bangs into the wall, knocking the print off axis - resulting in a fail. (I confirmed this by having my print fail right on the first support
[image]
 pass over the void.)

Ghostkeeper, 2020-04-06:
The support is printed separately before starting the main object. If you have two pieces of support, it prints those two pieces first on a layer before it starts printing the object. That's why you are seeing a travel move in the layers where it's printing support.

The main purpose of combing is to prevent going through the walls of your object. It doesn't go deciding that it needs to cross the wall to go inside your object and then cross the wall again to go outside it again in order to reach your second support piece. It doesn't need to cross any walls to reach the second piece of support, since it's already outside of any walls. It looks like it's behaving as expected in that screenshot.

GregValiant, 2024-11-14:
Hello.
Is "Avoid Supports when Traveling" as implemented in current Cura versions (5.8.0 and up) sufficient to close this?

closed 2024-12-10 by github-actions[bot]