Ultimaker/Cura #2402: support structure plugin

gokul2907, 2017-09-08:
Is there any plugin available to calculate the volume of support structure generated

fieldOfView, 2017-09-08:
No, there is not. Do you want the volume of the extruded material to print the support, or the volume of the entire support structure?

gokul2907, 2017-09-08:
Is it possible to get the volume of the extruded material to print the support structure .

fieldOfView, 2017-09-08:
If you print the support structure with a secondary extruder, the amount of material is shown in the interface (length and grams & price if you have entered the appropriate data for the materials).

Ghostkeeper, 2017-09-15:
Yes, but then the support might also be printed with different settings so you have to be careful about that.

A Cura plug-in for the amount of material in the support is fairly difficult to make because it involves parsing the g-code that we receive. A feature could be built in CuraEngine though that sends over the material usage per feature, so we won't have to calculate it in the front-end.

Ghostkeeper, 2018-02-12:
Our project manager refused this enhancement because it was not considered important enough.