Ultimaker/Cura #2403: Speeds are messed up

stema84, 2017-09-08:
I was using s3d, had a few issues with it so now I'm trying to use cura 2.7. The problem is that it's printing the first layer too fast IMO. In s3d printing speed was 50, and 1st layer 50%, in cure I set 1st layer to 25, but I feel that it does it faster which I don't really like, there are some issues. Any suggestions?
ps. Seems like every speed is messed up, travelling between elements REALLY slowly, than printing at the speed of traveling.. Using stable 2.7

smartavionics, 2017-09-09:
How about attaching a screengrab that shows the speed settings you are using and then maybe someone will be able to make a useful comment?

stema84, 2017-09-10:
you're right, sorry
[image]
I'll post the gcode soon

stema84, 2017-09-10:
here's the g-code, it has 3 kind of models in it (dial knobs, fists, graduation hats), whenever it has finished a layer of a graduation hat, it travelled to the next model extremely slowly on a wierd path, scratching the surface just printed. Then when it did the 1st outer line for one of the knobs that was too fast for my taste.
https://drive.google.com/open?id=0B_LDy_PeVVWmc0x6NG1VU29ldXc

smartavionics, 2017-09-10:
OK, thanks for the data. I can see from the data that you have found the same bug as https://github.com/Ultimaker/Cura/issues/2371 and a fix for that has already been submitted for inclusion in the next release.

stema84, 2017-09-10:
Thank you! It should've been a z-hop but actually it carved a bit into the layer so it haven't lift it (don't know if it lowered it or the carving was because of the heat)

smartavionics, 2017-09-10:
If your printer can handle a higher z speed than 5 mm/s, try entering the speed it can handle into the Maximum Z Speed setting. The problem is that it is using the speed used for the z-hop for the following travel and extrusion moves which is why they happen dog slow. Other than that, the only other option is to turn off the z-hop.

Ghostkeeper, 2017-09-15:
Duplicate of #2371.

closed 2017-09-15 by Ghostkeeper