Ultimaker/Cura #2404: Printer via Malyan not detected

ARandomGuyCalledMe, 2017-09-08:
I use Malyan to connect my Monoprice select mini to my pc
It worked with other Versions
Malyan creates a Port, the Port" COM100"

nallath, 2017-09-11:
I can't really help you with this, as we don't have monoprice printers. The USB printing isn't something that gets a whole lot of attention from us, as no UM printer really uses it.

Ghostkeeper, 2017-09-15:
When connecting via USB, Cura tries various baud rates and sends the text `M105` to it, which is g-code that requests the temperature. We then expect an answer that starts with `T:`, which indicates the current temperature.

I know of Malyan that they use a different standard than g-code, X3G. Perhaps that is the problem. In any case, fixing this would require a specialised version of the USB connection plug-in. We can't develop that because we don't have access to the printer.

closed 2018-03-02 by ChrisTerBeke