Ultimaker/Cura #2405: When using gradual infill bottom surface not max infill

mikeq3, 2017-09-09:
Gradual infill should start at the selected infill (say 30%) then thin out and then back to 30% for top surface. Not start thin on bottom and go to max on top. What about the strength of the bottom surface?

nallath, 2017-09-11:
We designed the gradual infill to not be about strength, but more about how well something is supported. You want a denser grid near the top surface, so you get less sagging. We haven't really noticed the need for denser support at the bottom of the support.

mikeq3, 2017-09-13:
That would fine for something that fits on a table and has a bottom to the model. But not for a piece like a tool or something functional that has no top or bottom. Can we make this an option?

nallath, 2017-09-14:
Some of the 3D infills do this way better. Concentric subdivision for instance.

mikeq3, 2017-09-14:
So can we have an option that has the infill go from say your selected fill drnsity to sparse and then back to selected infill? Instead of sparse to selected infill.

Ghostkeeper, 2017-09-20:
> That would fine for something that fits on a table and has a bottom to the model. But not for a piece like a tool or something functional that has no top or bottom.

There is always a top and bottom to your print unless you're printing in zero-G, because the sagging is not an issue of the model geometry but of the gravity while you're printing. Only the top skin gets pulled down, not the bottom skin.

There's something to be said for going the other way around because most of the print typically gets the sparsest infill layer which is, say, 1/8th of the density the user requested. So, for visuals, like, if people look in layer view. But the counterargument from the sagging's point of view is also very strong.

Phil1988, 2019-10-23:
+1 on this. 
A minimum and maximum infill should be selectable for better controling.

In many cases ~50% infill is more than enough for the top infill section.
But 50% is too much for the bottom part.

The gradual infill always makes the lower sections to half the density.

Lets say you print a 100mm height part.
What most people want is low density at the lower section and then ~50% (in many cases more like 30-40%) on the top section.

With the options right now you can either set the gratual infill to 2 steps with a height of 50mm.
This would result in much wasted filament in a 50mm section with 50%infill.
Lower half would be 25% infill which is ok...

Or you can set 3 steps with a height of 34mm.
This would still make a huge section with much wasted filament on the top section.
The middle would be ok with 25%.
Bottom would be pretty weak with 12.5% infill and causing the middle 25% infill to having problems not to sag and bind to the lower infill. This is more like bridging.

More steps is no option because this would result in almost no binding between the infill sections.

Best would be to have the option to define a minimum infill percentage and to define a top infill percentage and when the gradual infill should start (best also in percent).

This way you can set 20% to the bottom section and then start gradual infill in lets say 3 steps up to 50% starting at 90% height.
The model would then be printed with 20% infill for the first 90mm, then 35% for 5mm and then 50% for the last 5mm.

This model would be stronger, faster to print and also cheaper to print (less material and less power consumption because of lower print time).
There is no downside on this print compared to the ones you can do right now.

I hope you can include these settings to the slicer.
Right now the interest is not that much, because users are not familiar with that option.
This is mostly because slicers havent implemented this option at all or in a way its not usable.
You can be sure, that a option like I mentioned would be used by many users if they know this option exists.

Phil1988, 2019-10-24:
You mean "1 step" in this case.

Yes that would be possible. But it would be more controlable if you can just set the minimum infill as well.

There is nothing you can do to get 50% infill on top and 20% at the lower 95% or any other "perfect combination" for the model you are currently slicing with the infill you like.

This is like a workaround but nothing serious.

I get your point on the layer adhesion between the layers.
So it might be a good solution to have both gradually infill methods.
In some cases you need it, in some you dont.

Ghostkeeper, 2019-10-24:
I'm arguing that not doing 50% reductions at every step will be weaker in all cases, so you'll never need it. Including it would then just cause confusion.