Ultimaker/Cura #2406: How to get the development instance setup?

vivekanand1101, 2017-09-09:
I want to work on this s/w but, i am not able to see how i can setup development instance for the same. I run cura_app.py and it throws error about not able to import  UM. I am guessing UM is Uranium and i have it's sources downloaded as well. I don't know how i can get uranium to be installed for this.

fieldOfView, 2017-09-09:
Clone both Cura and Uranium in separate folders. Add the Uranium folder to the PYTHON_PATH env variable

vivekanand1101, 2017-09-09:
ok, that workd but, i don't have the plugins now

vivekanand1101, 2017-09-09:
It looks for plugin.json in plugin folders and it's not there. What should i do?

vivekanand1101, 2017-09-09:
It's actually there but, somehow it's failing to find: 
```
2017-09-09 19:33:03,793 - ERROR - UM.Logger.logException [76]: Exception: Unable to find the required plugin.json file for plugin ImageReader                            
2017-09-09 19:33:03,793 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):                                                                        
2017-09-09 19:33:03,794 - ERROR - UM.Logger.logException [80]:   File "/home/vivek/Uranium/UM/PluginRegistry.py", line 402, in _populateMetaData                         
2017-09-09 19:33:03,794 - ERROR - UM.Logger.logException [80]:     with open(metadata_file, "r") as f:                                                                   
2017-09-09 19:33:03,794 - ERROR - UM.Logger.logException [80]: FileNotFoundError: [Errno 2] No such file or directory: '/home/vivek/Cura/cura/../plugins/ImageReader/plugin.json' 
```

nallath, 2017-09-11:
If it's not there, you aren't working from master.

vivekanand1101, 2017-09-11:
great, thanks. I was on master but, it wasn't up to date.

However, i get this error now:
```
2017-09-11 13:20:42,680 - ERROR - UM.Logger.logException [76]: Exception: Could not deserialize container ultimaker3_extruder_left #2                                    
2017-09-11 13:20:42,767 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):                                                                        
2017-09-11 13:20:42,767 - ERROR - UM.Logger.logException [80]:   File "/home/vivek/Uranium/UM/Settings/ContainerRegistry.py", line 204, in load                          
2017-09-11 13:20:42,768 - ERROR - UM.Logger.logException [80]:     new_container.deserialize(f.read())                                                                   
2017-09-11 13:20:42,768 - ERROR - UM.Logger.logException [80]:   File "/home/vivek/Cura/cura/Settings/ExtruderStack.py", line 96, in deserialize                         
2017-09-11 13:20:42,769 - ERROR - UM.Logger.logException [80]:     super().deserialize(contents)                                                                         
2017-09-11 13:20:42,769 - ERROR - UM.Logger.logException [80]:   File "/home/vivek/Cura/cura/Settings/CuraContainerStack.py", line 352, in deserialize                   
2017-09-11 13:20:42,769 - ERROR - UM.Logger.logException [80]:     super().deserialize(contents)                                                                         
2017-09-11 13:20:42,769 - ERROR - UM.Logger.logException [80]:   File "/home/vivek/Uranium/UM/Settings/ContainerStack.py", line 357, in deserialize                      
2017-09-11 13:20:42,770 - ERROR - UM.Logger.logException [80]:     raise Exception("When trying to deserialize %s, we received an unknown ID (%s) for container" % (self.
_id, container_id))                                                                                                                                                      
2017-09-11 13:20:42,770 - ERROR - UM.Logger.logException [80]: Exception: When trying to deserialize ultimaker3_extruder_left #2, we received an unknown ID (generic_pla_
ultimaker3_AA_0.4) for container                                                                                                                                         
2017-09-11 13:20:42,774 - ERROR - UM.Logger.logException [76]: Exception: Could not deserialize container ultimaker3_extruder_right #2                                   
2017-09-11 13:20:42,774 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):                                                                        
2017-09-11 13:20:42,775 - ERROR - UM.Logger.logException [80]:   File "/home/vivek/Uranium/UM/Settings/ContainerRegistry.py", line 204, in load                          
2017-09-11 13:20:42,775 - ERROR - UM.Logger.logException [80]:     new_container.deserialize(f.read())                                                                   
2017-09-11 13:20:42,776 - ERROR - UM.Logger.logException [80]:   File "/home/vivek/Cura/cura/Settings/ExtruderStack.py", line 96, in deserialize                         
2017-09-11 13:20:42,776 - ERROR - UM.Logger.logException [80]:     super().deserialize(contents)                                                                         
2017-09-11 13:20:42,776 - ERROR - UM.Logger.logException [80]:   File "/home/vivek/Cura/cura/Settings/CuraContainerStack.py", line 352, in deserialize                   
2017-09-11 13:20:42,777 - ERROR - UM.Logger.logException [80]:     super().deserialize(contents)                                                                         
2017-09-11 13:20:42,777 - ERROR - UM.Logger.logException [80]:   File "/home/vivek/Uranium/UM/Settings/ContainerStack.py", line 357, in deserialize                      
2017-09-11 13:20:42,778 - ERROR - UM.Logger.logException [80]:     raise Exception("When trying to deserialize %s, we received an unknown ID (%s) for container" % (self._id, container_id))                       
2017-09-11 13:20:42,778 - ERROR - UM.Logger.logException [80]: Exception: When trying to deserialize ultimaker3_extruder_right #2, we received an unknown ID (generic_pla_ultimaker3_AA_0.4) for container    

2017-09-11 13:20:42,778 - DEBUG - UM.Settings.ContainerRegistry.load [214]: Loading data into container registry took 33.6339316368103 seconds                           
2017-09-11 13:20:42,856 - DEBUG - UM.Controller.setActiveView [84]: Setting active view to SolidView                                                                     
2017-09-11 13:20:44,432 - CRITICAL - cura.CrashHandler.show [39]: An uncaught exception has occurred!                                                                    
2017-09-11 13:20:44,468 - CRITICAL - cura.CrashHandler.show [42]: Traceback (most recent call last):                                                                     
2017-09-11 13:20:44,469 - CRITICAL - cura.CrashHandler.show [42]:   File "./cura_app.py", line 82, in <module>                                                           
2017-09-11 13:20:44,470 - CRITICAL - cura.CrashHandler.show [42]:     app.run()     
2017-09-11 13:20:44,471 - CRITICAL - cura.CrashHandler.show [42]:   File "/home/vivek/Cura/cura/CuraApplication.py", line 639, in run                                    
[... cut ...]
2017-09-11 13:20:44,482 - CRITICAL - cura.CrashHandler.show [42]:     used_extruders = extruder_manager.getUsedExtruderStacks()                                          
2017-09-11 13:20:44,482 - CRITICAL - cura.CrashHandler.show [42]:   File "/home/vivek/Cura/cura/Settings/ExtruderManager.py", line 471, in getUsedExtruderStacks         
2017-09-11 13:20:44,483 - CRITICAL - cura.CrashHandler.show [42]:     used_extruder_stack_ids.add(self.extruderIds[str(global_stack.getProperty("adhesion_extruder_nr", "value"))])                                
2017-09-11 13:20:44,483 - CRITICAL - cura.CrashHandler.show [42]: KeyError: '0'  
```

nallath, 2017-09-11:
You should probably delete your local files; Updating from master won't trigger the upgrade scripts, so now you have a bunch of data that it doesn't understand.

vivekanand1101, 2017-09-11:
do you mean deleting ~/.local dir? Or, you want me to delete and re-clone cura? I have tried the second one. In fact, i re-cloned Uranium as well. But, same error. I tried python3 setup.py build but, it seems it will only work on windows. Is that true?

nallath, 2017-09-11:
The cura local files in the ~/.local dir. The second one won't change anything.

vivekanand1101, 2017-09-11:
Great, Thanks @nallath @fieldOfView :)

closed 2017-09-11 by vivekanand1101