Ultimaker/Cura #2408: Adhesion options on per object settings

jtoves, 2017-09-10:
In 2.7, the bed adhesion options don't seem to be visible in the menu to select custom print settings per object. They are visible only in the overall config when "Show All" is selected, but only the top menu lable is visible when customizing print settings per object.

nallath, 2017-09-11:
Bed adhesion can only be set globally. You are right that there is an issue with the category not being hidden in that screen.

Ghostkeeper, 2017-09-15:
The behaviour is as intended. Basically, you can't have a different build plate temperature for every object because most (or all) printers don't support heating specific patches of the build plate differently, and g-code doesn't support it either.

jtoves, 2017-09-15:
With a project containing multiple objects being printed in PLA without a heated bed, that doesn't matter as much as the verticle stability I need that a brim adhesion method provides. What does matter is access to printing a brim for one of the objects on the bed which requires a brim for stability, not a different bed temperature. If the drim feature was accessible per object, I could print use the setting to stabilize the one object that needs it instead of having to setup a separate print run.

Ghostkeeper, 2017-09-20:
Sorry, I misread this issue as having to do with bed temperature, instead of bed adhesion.

Ghostkeeper, 2017-09-20:
I don't see the bed adhesion setting itself becoming per-object any time soon, because of the way in which rafts are implemented right now.

Brim line count can be done though. It's currently implemented by taking the entire bottom layer (of all models and extruders) and generating lines around that polygon. Instead we could create offsets for each object individually. However you then need to do a quadratic number of polygon intersections, quadratic to the number of objects. This is because each line of each brim may intersect with each line of each other brim and we need to prevent overextrusion as well as keep the brim lines parallel with the adjacent objects. I'm not sure if that is feasible.

ianpaschal, 2018-03-05:
Given feasibility issues and ticket age, I'm closing this. Obviously we can reopen if it we're readdressing per-object-settings in the future but I don't think warrants the "needs to be fixed" nature of a `Deferred` label.

closed 2018-03-05 by ianpaschal