Ultimaker/Cura #2409: Cura 2.7.0 - Splash screen hanging for > 2 min after loading interface

dennisdegryse, 2017-09-10:
I updated from Cura 2.7.0 beta to Cura 2.7.0 and noticed that the splash screen is hanging on the "Loading interface" step. After checking the logs I read the following warning (previous and next DEBUG line supplied for clarity):

```
2017-09-10 19:48:32,703 - DEBUG - UM.Qt.Bindings.Theme.load [232]: Loaded theme C:\Program Files\Cura 2.7\resources\themes\cura
2017-09-10 20:01:16,426 - WARNING - cura.Settings.ContainerManager.getContainerMetaDataEntry [184]: Could not get metadata of container  because it was not found.
2017-09-10 20:01:17,652 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction._createAdditionalComponentsView [216]: Creating additional ui components for OctoPrint-connected printers.
```

At first I thought it just hang indefinitely, so I tried the common cleaning advice (AppData subdirectories and reïnstall). This didn't solve the hang, so I let it load a bit longer. The interface loaded and I captured the WARNING line above, hoping the container load time can be fixed. 

If the step is not crucial (I assume it isn't ,given it's just a warning and Cura was able to start), is there some theory on why the Uranium container ID is empty/undefined and why it is still querying on an empty argument?

Ghostkeeper, 2017-09-15:
The step is not crucial. I have it too but for me it doesn't hang the interface. The "Loading interface" step takes a second or two for me. The "Loading machines" step takes much longer. I don't know what causes this empty query but it doesn't seem to be causing problems now.

I'd like to shift the blame to the OctoPrintPlugin instead but I don't know if that is causing the problem either, of course.

For me this bug doesn't reproduce :(

dennisdegryse, 2017-09-15:
I restarted Cura today and didn't experience the delay. It's probably not necessarily a Cura issue. I'll see if it comes back and if so, I'll try to bypass it in code and test the outcome. So the issue should probably left on hold or even closed :) Could you provide me some insight on what metadata it's loading (the functional meaning of it)? Perhaps the second parameter could also be logged (I only get to see the empty ID).

The loading machines step is also a bit slow, but not really bothering. I assume it's iterating some ports (haven't dug into the code for that yet) and therefore mostly depends on the IO of the host (understandable). 

Also a practical question: Do you have a shared CI set up to compile a Cura branch for Windows? I do dev in linux, but use Cura on Windows and I'm not planning on setting up the build env on Windows. 

Thank you!

Ghostkeeper, 2017-09-24:
It's hard to tell what metadata is being loaded, but it is metadata of a container, so that is either a profile, a material or a printer. It's probably something that has to be displayed in the interface as well, seeing as it happens while the interface is being built. So probably something like the current profile name, or current printer name.

The loading machines step is slow mostly because it's unpacking the materials. It's not I/O bound. The thing is that the materials can have different settings for each of the printers it supports. So internally we build a profile called "generic_pla_ultimaker_2_plus", a profile called "generic_pla_prusa_i3_mk2", and so on. Also, some printers have several variants and these can have separate profiles in a material as well. So take the 82 materials and multiply that with 80 printers and all of their variants/nozzles and run that in Python. What you get is a slow loading screen :wink: It's basically an issue of our architecture right now. We load a lot of things lazily, but even the metadata loading is taking a lot of time right now.

Yes, we have a CI system set up to test Cura on Windows and Linux. Not on MacOS yet (though we have a build system on MacOS that makes daily builds).
You can see the CI system here: http://hal9000.ultimaker.com:29130/blue/organizations/jenkins/Ultimaker%2FCura/activity
When a pull request is created, the unit tests on that branch are automatically executed, so you can see the result of the unit tests you've made.

Currently the unit tests seem to be failing on CI...

ChrisTerBeke, 2017-09-28:
Closing since original reported cannot reproduce anymore. Let's re-open if further problems arise.

closed 2017-09-28 by ChrisTerBeke