Ultimaker/Cura #2410: Cura Update Deleted My Second Machine Profile

hashtagdavi, 2017-09-10:
My Cura updated on my machine today. i have two printer profiles saved, as i have two different 3D printers. My second printer profile is gone, upon opening the application with the new update prompt. I now need to re input the profile for the printer, however i believe this is not supposed to happen.
:-/

nallath, 2017-09-11:
Could you share your logs?

hashtagdavi, 2017-09-11:
Will do

Ghostkeeper, 2017-09-18:
The log of the first time you started 2.7 should tell something if it went wrong. It would also be useful to have the plug-in file (or entire folder) that went missing.

hashtagdavi, 2018-01-01:
i apologize, i have been trying to share the logs for yall. i cant for the life of me figure out how to do so with cura. i dont know what im doing wrong, and i promise im not a noobie i swear lol. i am honestly embarassed i could not figure this out

hashtagdavi, 2018-01-01:
also, as many updates have been pushed now, im not sure how relevant this still is? 

i realize now , i am running 15.02.1. on the machine i had the problem with. updating should fix my problem im sure. i notice though the version is so old it states there is no update. 

though i have 3.1 on my other machine. and setting up my two printer profiles on here have been very different,

the differences are vast....
the first machine is easy,  a Printrbot Simple Metal with heated bed. the original machine from 2014. the process was easy as Cura has the printer already listed with accurate settings as well. 

the next machine has been a problem for me. 
As setting up the seccond, it is a custom required, as the printer is an "Overlord Pro" and is not listed in the inventory of recognized printers to Cura.   

the issue i have is that the printer require's G Code Flavor : "UltiGCode" for the machine, 
and this is no longer listed as a flavor option with 3.1 Cura. unless it is under a different name, i searched online and on forums and could not find the solution for requiring G Code flavor "UltiGCode" in a custom printer profile. 

i have yet to try and print, though i thought, since the Ultimaker 2 is listed, and this machine used UltiGCode as its flavor, that this would allow the printer to work correctly. i hope what i have done  is correct and wil work, as i am about to try printing with the printer and this new profile. 

ill post back here with results.

i apologize for the logs and the original problem. im sorry if i have caused any inconvenience to your time.  

i would really appreciate your help and attention with this new issue that has come up for me, the G Code Flavor problem for my printer. Thank you for your time and i will report back here shortly.

Ghostkeeper, 2018-01-17:
Here's some information on where your log files are: https://github.com/Ultimaker/Cura#logging-issues

Ghostkeeper, 2018-01-17:
UltiGCode was renamed to "Ultimaker 2". That's the only printer we know of that uses UltiGCode. UltiGCode was developed in-house by Ultimaker and abandoned again for the next generation.

ianpaschal, 2018-02-27:
Since we haven't heard back from OP in more than a month with the required info I'm closing this.

closed 2018-02-27 by ianpaschal