Ultimaker/Cura #2411: 2.7.0 Printbed not in Picture

SirWinni, 2017-09-10:


SirWinni, 2017-09-10:
Wenn aus Versehen das Druckbett aus dem Monitorbild verschwindet ist es quasi nicht mehr auffindbar.
Es fehlt eine Funktion um das Druckbett im Bild zu zentrieren.
Außerdem ist es ziemlich nervig, dass beim heranzoomen das Objekt nicht in der Bildmitte bleibt, sondern das Bild nach irgendwo verlässt.

Ghostkeeper, 2017-09-15:
> When you accidentally lose the print bed from the view, it can sometimes not be found any more. There is no functionality to centre the print bed in the view.
Also, it is quite annoying that when zooming the object doesn't stay in the middle of the view, but the view goes somewhere else.

There is indeed no functionality to find the build volume again if you lost it in the view. You can usually find it by looking around, unless you zoomed out really far, but a function to go back to the default zoom and camera position could be useful.

There is a centre point for the camera, by default in the centre of the build volume. If the object isn't there at the centre, it won't zoom towards the object. You can "Move camera to selected object" in the preferences though, but by default this function is disabled. There is nothing more we can really do. Especially when there's multiple objects on the build plate.

fieldOfView, 2017-09-15:
A "Reset camera position" menu item could be a fairly trivial solution. I'll see what I can drum up.

PR #2439 (2017-09-19): Add menu item to reset the camera
This PR adds a menu item to the view menu to reset the camera position and lookat, homing the camera. This can be used when the user "looses" the buildplate from the view.

Fixes #2411 and contributes to #2361
closed 2017-09-19 by nallath