Ultimaker/Cura #2412: Infill "shift"

darkdvd, 2017-09-11:
For a specific project, I need to print an object that fits in a volume of 210.8 x 200.2 x 195.5 mm.
For this project, I use an infill of 3% (triangles)
I need to position the fill on the axis of the object to reinforce some parts.
Problem: I can't move the object sideways to make it coincide with the infill.
Can you introduce the possibility of moving the X/Y origin of the infill?
[image]

darkdvd, 2017-09-12:
To begin, please excuse my English (I use GG translate)

In most cases, Infill is used only as an internal support for an object.

In my case, the Infill is an important mechanical element playing an essential role in the construction of an object.

With CURA, the Infill is positioned in the center of the buildplate, which is not a problem for most objects. : To adjust the position of the Infill, simply move the object on the board.

I am currently working on a project where the filling has to be done in the longitudinal axis of the object.

This object is inclined by 42 ° on the plate and it is 195.5 mm high.
I used the "Infill Line Directions" function to orient the Infill in the axis of my object using the values ​​[47,107,167] for an Infill in triangles.

Problem: I can not move my object to align the Infill longitudinally.

Could you add an option "Infill Origin" with two options in the parameters:
- Buildplate Origin
- Object Origin

Another parameter "Infill Origin Directions" would allow this position to be adjusted. Ex. [-5,15] to move the Infill from -5mm to X and 15mm to Y

Merci !

David.

smartavionics, 2017-09-12:
Err, it might be easier if you got yourself a bigger printer (just kidding).

amagr0, 2017-09-13:
That's a very important future feature in order to guarantee similar mechanical properties in the same models that are placed in different positions on the print bed.
Like the zscar that now have the option to "Z Seam Relative", a similar approch could be implemented to the infill.

[image]

smartavionics, 2017-09-13:
I agree, having the ability to move the infill origin so as to achieve uniform mechanical properties across multiple parts would be very good. I'm not familiar with the Cura infill code but I when I have some time I will look into how that could be achieved.

darkdvd, 2017-09-13:
Daid, Jun 4, 2014 - 12:17 PM
http://ultimaker.com/en/community/6937-ability-to-shift-infill-pattern-in-cura#reply-55665

darkdvd, 2017-09-13:
note : concentric infill is centered to the object.
https://github.com/Ultimaker/CuraEngine/blob/master/src/infill.cpp LINE 114

DDDirk, 2017-09-16:
I agree, I also once spent some time shifting a model around until I finally got the infill-walls where I wanted them. But I also had to play with the infill percentage to get something that seemed good. And it was actually rather a coincidence that I got the infill walls exactly where I wanted them, in different parts of the model (what worked in one location also worked in another).

Another way to achieve this is by incorporating the internal supporting structures in the model. That way you can make less homogeneous supporting structures, exactly where you want them. This is an as yet little used advantage of 3D printing, the ability to make internal structures just as easily as the visible external parts.
One way to do this is to make big holes in the model (where you don't want or need any material) and use 100% infill. Another is to place very thin bars (where you want the supporting structures) with 0% infill. (Or any percentage you like of course.) Both work with the boolean subtract of a CAD program. And in Cura, make sure 'mesh fixes > union overlapping volumes' is enabled.
The second method is a bit more complicated because the thin bars have to be sufficiently close to the shell, so they touch (which depends on the shell width). What happens is that Cura will create a shell around those thin bars, which then becomes the supporting structure. That's why they can be very thin (although they don't have to be).
Of course the bars have to be placed such that they don't need any support (eg not horizontal). Or with the second method you could use infill anyway, which then acts as a support structure (just like when the model has a horizontal top). And then it would be nice if you could get that where you want it, as requested above.
Note that if you use infill, those internal shells will be printed at shell speed, which will likely be lower than infill speed. And thus possibly stronger.

However, this is more work, possibly too much if you want to do a single print. And for smaller models it might not work well. So I am still in favour of the proposition.

smartavionics, 2017-09-17:
Hi Folks, I've done a bit of work on this and have basic "infill origin" support working. Have a look at this image, there's two parts, the infill is identical for the two parts and has been offset  by 1mm to the right and 4mm upwards so that one of the infill "nodes" coincides with the hole on the bottom left of each part. If you look at the settings you can see new Infill X Offset and Infill Y Offset fields. The infill does not shift when the part it shifted on the build plate.
[image]
Is this the sort of capability you are looking for?

darkdvd, 2017-09-18:
Yes ! 
And if you change the infill direction to "[30,90,150]" the Infill for the holes on the bottom will be aligned too.
Just play with the infill % and tada !

darkdvd, 2017-09-18:
The infill shift is more useful when you cannot move the piece on the buildplate because it is too large.

darkdvd, 2017-09-18:
smartavionics : is the infill a per model setting too ?

smartavionics, 2017-09-18:
> And if you change the infill direction to "[30,90,150]" the Infill for the holes on the bottom will be aligned too.
Just play with the infill % and tada !

Like this?
[image]

smartavionics, 2017-09-18:
Yes, the offsets are settable per-model.

darkdvd, 2017-09-18:
Definitely to put in the next version of CURA !

Labern2, 2017-09-18:
+1 this will be very handy.

Ghostkeeper, 2017-11-26:
This was implemented now. I'm not sure if this got in time for 3.0, but it's in 3.1 anyhow.

closed 2017-11-26 by Ghostkeeper