Ultimaker/Cura #2414: Print time estimation incorrect

modem7, 2017-09-11:
Hey guys,

Something I've thought about when thinking about the disparity between print time estimation from cura and the actual print time.

Would it make sense, in printer settings under maybe an "optional" or "advanced" section, to have the firmware printer acceleration and jerk, especially as cura does it's best to estimate, but doesn't have those two crucial pieces of information. 

This way cura can better estimate ETC and also as an added bonus, the jerk and acceleration boxes could take those as the "default" values, and users can modify them in software if they so wish, thus overriding what is set in the printer.

Thoughts?

nallath, 2017-09-12:
Interesting notion, but all our printers accept acceleration commands when given, so the settings set in Cura are the ones actually used. It might be that it's different from other printers, but as in all other cases; We won't spend time developing features that UM doesn't need. If anyone feels like building it, we do welcome pull requests.

modem7, 2017-09-12:
This isn't for controlling anything additional. This is to get your print
time estimation far more accurate than it is now (I've had 1-2 hours in
difference).

You already have print jerk and acceleration settings within cura, so there
is no need to replicate that, this is purely for ETC/ETA calculations which
are..... Not accurate due to an assumed value.

Thisismydigitalself, 2017-09-12:
If this sets M205 command that overrides Marlin's eeprom settings then this might just be the cause for the time estimation differences. i will give this a try.

nallath, 2017-09-12:
Cura actually sends the acceleration values to the printer. Same goes for jerk. This is why (atleast for UM printers) the time estimations are pretty close to reality. I'm not sure why this isn't working for other printers.

modem7, 2017-09-12:
Surely it would only send those values if jerk and acceleration settings
are enabled.

In my particular case (using a wanhao di3 2.1 connected via octoprint),
I've done my calibration and stored it in eeprom, so I wouldn't enable jerk
and acceleration control via the software as it's being dealt with via the
firmware.

My original post was to state: as I know the values of jerk and
acceleration, I could input that to a box in printer settings so that ETC
can be better computed, thus allowing a nice hybrid between pretty good ETC
and firmware accel and jerk control

nallath, 2017-09-12:
So the default values need to be updated? That can be done; just change the printer definition.

modem7, 2017-09-12:
Ah really?

Not at my computer at the moment, but I can't recall an area in custom
printer where I can update the default values of jerk/accel?

If I'm being blind, then awesome!

nallath, 2017-09-12:
Custom printer doesn't have an area for it, but it's up to the printer definition (in this case the wanhao definition) to set correct default values.

modem7, 2017-09-12:
Would it be worthwhile in the custom printer section to include an area
where people with custom printers (e.g. Corexy) to be able to input those
values?

nallath, 2017-09-12:
Yea, but our priority is on fixing issues / features that influence how our machine behaves. As this isn't something that does that, it's not likely that we will actually build it.

modem7, 2017-09-12:
That's a fair comment.

Is there a location/module within cura that you guys use to set the default
accel/jerk settings for inbuilt printer profiles?

fieldOfView, 2017-09-12:
It would probably go in Machine Settings.

As a stop-gap meassure, your configuration folder has a folder in it named "definition_changes". In it, there's an ini file for your printer. Under [values], you can add lines such as:
```
machine_acceleration = 4000
machine_max_acceleration_x = 9000
machine_max_acceleration_y = 9000
machine_max_acceleration_z = 10000
```
etc

Ghostkeeper, 2017-09-20:
I don't think we should expose those machine settings to the user. Just having them as machine settings is fine; leave it up to the creator of the definition to set them correctly.

closed 2017-09-20 by Ghostkeeper