Ultimaker/Cura #2419: Feature request: Retraction Extra Prime Amout - negative values

amagr0, 2017-09-12:
Since I did not found anymore the previous post where this feature was already requested, I'm opening a new issue with the same request.

**Material:**

- Retraction Extra Prime Amout - positive and negative values

and also

**Experimental:**

- Coasting Extra Prime Amout - positive and negative values

_

This settings could improve a lot the problems caused by the excess material in the z-scar when printing with thick layers.

konvoj, 2017-09-12:
The first part is solved by [#2392](https://github.com/Ultimaker/Cura/issues/2392).

amagr0, 2017-09-12:
Tks konvoj

Ghostkeeper, 2017-09-20:
Um, there is no Coasting Extra Prime Amount setting. There's only the coasting volume, coasting minimum volume and coasting speed (and enable coasting).

Ghostkeeper, 2017-09-20:
Did you intend for such a setting to be created?

How would that work? While coasting you still extrude a bit or retract a bit? Currently it's changing a segment of the path into a travel move, so it's extruding 0.

89AMStudio, 2017-09-24:
konvoj how to use the fix, that is the CuraEngine this is Cura, how can we update that?
We only have .pyd files in our Cura 2.7 folders and that fix is on a .cpp that we do not have.

Ghostkeeper, 2017-09-28:
To get the fix you'd need to get a newer version of CuraEngine. You would either need to compile it yourself which requires some knowledge about CMake and C++ build systems, or you'd need to wait for a release of Cura. It'll be in the next release, which is due around halfway through October.

ChrisTerBeke, 2017-11-14:
3.0 and higher have this.

closed 2017-11-14 by ChrisTerBeke

89AMStudio, 2017-11-14:
I have 3.0.4 and this still doesn't work.

Try to set Retraction Extra Prime Amout to 0 and save the gcode file.
Try now to set it negative and compare the 2 gcodes.

They are exactly the same.

ChrisTerBeke, 2017-11-14:
I'm getting differences when I do that, for example the 'material used' text at the top of the G-code shows that there's a difference in filament used. Diff here: http://www.mergely.com/8VqTZ9v0/.

89AMStudio, 2017-11-14:
Yeah sorry I have checked again and this time seems to work.

Sorry to bother for nothing, have a nice day/evening!

odospace, 2018-01-06:
I can't add a negative amount in Cura 3.1.0?

Ghostkeeper, 2018-01-23:
You can't?

[image]

Still seems to work.

devils27, 2018-02-27:
It doesnt work, it lets you input the number but if you look at the gcode it doesnt do anything.  The retract amount/length is the same as the prime amount/length

Ghostkeeper, 2018-03-21:
Hmm, yeah you're sort of right there. It does do something though. When I set it to -100 then it seemed to disable most of my retractions, turning them into non-retracted moves. But this behaviour is not correct obviously.