Ultimaker/Cura #2420: Cura does not detect a Monoprice Maker Select Plus over USB.

pbannister, 2017-09-13:
My printer is a Monoprice Maker Select Plus, which is a re-branded Wanhao Duplicator i3 Plus (which is in turn a cheap Chinese clone of the early Prusa i3). Cura 2.7 does not detect my printer when connected via USB.

I am not asking you to fix my problem. As a software guy, I am perfectly able to write and test a fix. This issue is mainly to track the problem. 

Though I would welcome a general hint about where to look in the sources. :)

nallath, 2017-09-13:
Nice! You need to look at the "USB printing" plugin. All the code regarding USB connection is there.

If you have any further questions, feel free to ask them.

Ghostkeeper, 2017-09-20:
Specifically: https://github.com/Ultimaker/Cura/blob/3cdf3f609779db443d01decfd9903e2a0c168d5b/plugins/USBPrinting/USBPrinterOutputDevice.py#L310

joelisjoel, 2018-01-29:
Having a similar problem with a Monoprice mini select v2.

Some part of the connection is detected when the USB cable is attached, but baudrate detection fails. 

Is this a duplicate bug? Is there a way to troubleshoot this?

> 2018-01-28 19:48:19,041 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [328]: Attempting to connect to printer with serial /dev/cu.usbmodem1411 on baud rate 57600
> 2018-01-28 19:48:43,545 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [328]: Attempting to connect to printer with serial /dev/cu.usbmodem1411 on baud rate 38400
> 2018-01-28 19:49:08,049 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [328]: Attempting to connect to printer with serial /dev/cu.usbmodem1411 on baud rate 19200
> 2018-01-28 19:49:32,552 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [328]: Attempting to connect to printer with serial /dev/cu.usbmodem1411 on baud rate 9600
> 2018-01-28 19:49:57,056 - ERROR - USBPrinting.USBPrinterOutputDevice._connect [366]: Baud rate detection for /dev/cu.usbmodem1411 failed
> 2018-01-28 19:49:57,057 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [380]: Closing the USB printer connection.
> 2018-01-28 19:49:57,058 - DEBUG - USBPrinting.USBPrinterOutputDevice.close [385]: PrinterConnection.close: cannot join current thread (expected)
> 2018-01-28 19:49:57,066 - WARNING - UM.OutputDevice.OutputDeviceManager.removeOutputDevice [126]: Could not find output device with id /dev/cu.usbmodem1411 to remove

jackha, 2018-08-16:
@joelisjoel Can you try again? There have been some improvements regarding timings and retries.

pbannister, 2018-08-17:
Ah. Guess I should add some information.

First, it seems that MacOS silently detects and ignores USB devices, if not approved by Apple. The OS knows the device is there, and knows there is no installed driver, but says nothing to the user. This did not match my expectations. I expected some sort of notification, if a new driver was needed.

Once I installed the USB drivers (to match the control board in the printer), the printer became visible.

That said, I no longer attempt to use USB. On each printer, I have a Raspberry Pi running OctoPrint. I now use OctoPrint to control my printers, and Cura works well with OctoPrint. (Of course, Linux works without trouble, and recognizes the USB device.)

jackha, 2018-08-20:
Ahh thanks for the explanation! Then I'll close the issue. I didn't know this from OSX. Glad you found a workaround! This way your computer is free for other stuff as well :-)

closed 2018-08-20 by jackha