Ultimaker/Cura #2426: Feature Request - Z Hop only on outer shell/top

Eganwp, 2017-09-13:


Eganwp, 2017-09-13:
Final product print quality could be HUGELY improved if there was an option to enable Z-hop on the outer shell/top. Then we wouldn't have these great big nozzle travel lines showing on our finished prints. I don't want to enable Z-hop for the entire print job because of the zits it leaves on the outter shell. This would give everyone a vastly improved final product for print quality!

konvoj, 2017-09-14:
Have you tried the new experimental ironing feature in 2.7? When tuned right, this will give you the best looking top surface currently possible.
I agree however that having a Z-hop on top layer is a logical proposal. I could even imagine this to be a default setting.

Ghostkeeper, 2018-04-12:
This one was older than #3115, but the other one has more information now so let's close this and continue discussion over there.

closed 2018-04-12 by Ghostkeeper