Ultimaker/Cura #2427: Unable to slice error w/o further details

thesec, 2017-09-13:
Cura 2.7.0 / Mac OS shows me "Unable to slice with the current settings. the following settings have errors:" The dialog ends here, there are no settings highlighted in the "Print Setup" (which would be a nice feature btw :) ) nor seems there to be any other source to gather the failing settings from?!

:/

nallath, 2017-09-14:
Could you share your logs and your project file?

thesec, 2017-09-18:
Where to find the logs?
Am able to reproduce it when changing the layer height from .2 to .2x.

Am using the profiles provided in https://drive.google.com/file/d/0BwSHq9qaQ0euSmJKRElfM1NvWjQ/view.

Ghostkeeper, 2017-09-24:
Wow, that printer has a whopping 1.8MB of quality profiles in there.

Ghostkeeper, 2017-09-24:
It doesn't reproduce for me. I'm loading the Sigma printer in, at 0.4mm Brass, PLA, Draft Quality. Then changed the layer height to 0.23mm. I had a small cube loaded, and that sliced fine.

Ghostkeeper, 2017-09-24:
In Cura, you can go to Help -> Show Configuration Folder to get to the directory where the logs are located. Alternatively, you can navigate to `$User/Library/Application Support/cura/<Cura version>/cura.log`.

Nallath also asked for a project file. That could help. You can create a project file in Cura by going to File -> Save project. This should save all of your settings, your printer, profiles, and your model. Save it while you're having the error. Put it in a .zip and upload it here on Github. Maybe we can reproduce the issue then.

thesec, 2017-11-12:
Just happened again, chaning only the layer height from .2 to .25mm ...

Does the log reveal to you what (might have) happened?
[cura.log.zip](https://github.com/Ultimaker/Cura/files/1464881/cura.log.zip)

The error did not happen with other models and the very same model reloaded.

OT: 
2017-11-12 17:35:14,704 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] [WARNING] Couldn't load extruder.def.json file for extruder 1. Index out of bounds.
What might cause such? Does not show in the UI :/

Ghostkeeper, 2017-11-20:
This seems to be the relevant error in that log:

    2017-11-12 17:35:19,498 - WARNING - CuraEngineBackend.StartSliceJob._checkStackForErrors [74]: Setting infill_sparse_thickness is not valid, but ValidatorState.MinimumError. Aborting slicing.

Your Infill Layer Thickness setting is less than 1 in a per-object setting?

Ghostkeeper, 2017-11-20:
Thanks to @thesec's log I was able to make a fix for this bug: https://github.com/Ultimaker/Cura/commit/6d2bcd9b3e6acc7ae690c18758e9f6b6dd8beabb

Adding logs helps, folks :smile:

closed 2017-11-20 by ChrisTerBeke

thesec, 2017-11-20:
@Ghostkeeper yay - glad I could help and thanks for your swift support!

> Your Infill Layer Thickness setting is less than 1 layer in a per-object setting?
The layer infill in the project is set to .2mm but with the first extruder and not per object.

thesec, 2017-11-20:
@Ghostkeeper just re-read the description and it seems I might have been mislead by the "mm" label in the layer thickness input. If the input indeed is rather a factor (is it?) the label should imo not be "mm" but "factor" or "x" ... or have the entry in percent plus "%" to keep it consistent with other.

Ghostkeeper, 2017-11-27:
The setting is intended to sound like millimetres to the user. In that text box you enter some mm of layer thickness.

Under water this is converted to a number of layers that it must combine together. This is why we can't handle a value that's lower than one layer height. The setting should become red if you try that. There's also a hard maximum of 8 layers because of implementation limitations.

commit 6d2bcd9b (2017-11-20): Fix error message when slicing with per-object setting errors

The message was generating a list of settings that had an error state by going through all extruder stacks and the global stack, but didn't bother to check the per-object stacks. I could've added it to the regular message but then the user would be confused because he can't find any errors either. So instead I opted to specify that it happened in per-model settings. It's not perfect, but should narrow down the user's search considerably.

Fixes #2427.